balena-io/etcher #2129: Make hard drive writes `unsafe`

ladyada, 2018-03-21:
- Etcher 1.3.1
- Windows 7 64 pro
- Raspbian

Heya folks! I noticed in a recent update y'all now have external hard drives show up as options for burning images? (at least, im pretty sure this is recent?)

[image]

One of the things I really liked about Etcher is that it, historically, it made it really  hard for people to accidentally overwrite harddrives - compared to `dd` or WinDiskImager...

I'd like to suggest allowing-writes-to-non-SD-cards part of an 'unsafe' option? Nothing worse than realizing you just zorked a disk drive - by the time you notice, the FAT's gone :(

jhermsmeier, 2018-03-21:
Hey, yeah – the drive detection had quite some changes done in 1.3.1 to more reliably detect & filter devices by bus type, but that also introduced what you're describing here (since both are connected via USB and hence not classified as internal storage anymore). We've already defused this in https://github.com/resin-io/etcher/pull/2045, which will show warnings when selecting those devices and also prevent them from being automatically selected.

Also relevant https://github.com/resin-io/etcher/issues/2071

ladyada, 2018-03-21:
ooh shoot i didnt see https://github.com/resin-io/etcher/issues/2071 (i swear i looked! :)
closing with deference to #2071

closed 2018-03-21 by ladyada