balena-io/etcher #2139: EINVAL: invalid argument, write

mMerlin, 2018-03-24:
- Etcher cli 1.3.1
- Fedora 27 x86_64
- 2018-03-13-raspbian-stretch.zip 
- ?no access to devtools from shell?

```
sudo -i
cd «path/to/etcher»
export ETCHER_CLI_DEBUG=YES
./etcher --drive /dev/sdb ../../../Downloads/2018-03-13-raspbian-stretch.zip 
? This will erase the selected drive. Are you sure? Yes
EINVAL: invalid argument, write

Error: EINVAL: invalid argument, write
    at Error (native)
```

AppImage worked, but wanted to explore using the cli.  First, AppImage did not need elevated privileges to flash the image.  cli did.  Then this error.  Setting the debug flag did not get much extra information.

From the timing, the error occurs before any `real` write to the device.

lurch, 2018-03-24:
Duplicate of #2074 ?

mMerlin, 2018-03-24:
Looks like.  My search should have found that one.  Seem to have messed up search on the closed issues.

Looks like I need to build from repo if I want to continue playing with this.

lurch, 2018-03-24:
...or you could wait for #2138 ?

mMerlin, 2018-03-25:
I saw that, but do know know how soon it will be available, or even if it covers the problem.  Nothing I see in the information there mentions the related issue number, or the flags that seem to be the cause of the problem.  And since it breaks another closed issue, it becomes a choice on which piece to break.  *ALL* validation, or everything on only a few systems.  And maybe only when using the CLI as well, since the gui worked for me for exactly the same system, source file, and target device.

jhermsmeier, 2018-03-26:
@mMerlin the flags are mentioned in there, and more specifically in https://github.com/resin-io/etcher/pull/2055 – it's O_DIRECT and O_EXCL; and it doesn't necessarily break validation – that depends somewhat on the particular OS / kernel version and some other not quite nailed down factors. Etcher v1.4.0 should drop this week.

jhermsmeier, 2018-03-26:
Duplicate of #2074

jhermsmeier, 2018-03-26:
Closing this as duplicate, and since it's already been fixed

closed 2018-03-26 by jhermsmeier