balena-io/etcher #2142: etcher crash on install in win 8.1

dpap, 2018-03-24:
- **Etcher version:** 
1.3.1
- **Operating system and architecture:** 
Windows 8.1
- **Image flashed:** 
Check below 
- **Do you see any meaningful error information in the DevTools?** 

 Problem Event Name:	APPCRASH
  Application Name:	Etcher-Setup-1.3.1-x64.exe
  Application Version:	1.3.1.0
  Application Timestamp:	584dca80
  Fault Module Name:	StackHash_c1ea
  Fault Module Version:	6.3.9600.18895
  Fault Module Timestamp:	5a4b127e
  Exception Code:	c0000374
  Exception Offset:	PCH_B2_FROM_ntdll+0x0003C8AC
  OS Version:	6.3.9600.2.0.0.256.48
  Locale ID:	1032
  Additional Information 1:	c1ea
  Additional Information 2:	c1ead2bddb361611c10b6526705c0c31
  Additional Information 3:	7df1
  Additional Information 4:	7df17d4cd30346fddbfc9fd38feb160b

jhermsmeier, 2018-04-06:
@dpap so the installer itself is crashing?

dpap, 2018-04-09:
@jhermsmeier  yes the installer crashes and doesn't install the app