balena-io/etcher #2144: Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease

sherminator666, 2018-03-26:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

jhermsmeier, 2018-03-26:
Can you post how exactly you installed it, meaning which commands you ran?

sherminator666, 2018-03-26:
i did this
echo "deb https://dl.bintray.com/resin-io/debian stable etcher" | sudo tee /etc/apt/sources.list.d/etcher.list
and then did 
sudo apt-get update
it was then the mentioned error showed..

jhermsmeier, 2018-03-26:
Ref #1638, #1634, #1633

jhermsmeier, 2018-03-26:
Thanks, we'll look into it!

sherminator666, 2018-03-26:
also did this before update command
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61

i was just following instructions on https://github.com/resin-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

lurch, 2018-03-26:
Hmmm, this still works for me (on Ubuntu 14.04)
```bash
$ sudo apt-get update|grep bintray
Get:5 https://dl.bintray.com stable InRelease
Ign https://dl.bintray.com stable InRelease
Hit https://dl.bintray.com stable Release.gpg
Hit https://dl.bintray.com stable Release
Hit https://dl.bintray.com stable/etcher amd64 Packages
Hit https://dl.bintray.com stable/etcher i386 Packages
Get:11 https://dl.bintray.com stable/etcher Translation-en_GB
Get:13 https://dl.bintray.com stable/etcher Translation-en
Get:16 https://dl.bintray.com stable/etcher Translation-en_GB
Get:17 https://dl.bintray.com stable/etcher Translation-en
Get:18 https://dl.bintray.com stable/etcher Translation-en_GB
Get:19 https://dl.bintray.com stable/etcher Translation-en
Get:20 https://dl.bintray.com stable/etcher Translation-en_GB
Get:21 https://dl.bintray.com stable/etcher Translation-en
Get:22 https://dl.bintray.com stable/etcher Translation-en_GB
Ign https://dl.bintray.com stable/etcher Translation-en_GB
Get:23 https://dl.bintray.com stable/etcher Translation-en
Ign https://dl.bintray.com stable/etcher Translation-en
```

Perhaps the `apt-get` command works slightly differently on Debian 9?

Hmmm, did a bit of searching and eventually found https://unix.stackexchange.com/questions/387053/debian-9-apt-and-gpg-error-inrelease-the-following-signatures-were-inva

andersab, 2018-05-02:
Didn't work for me either...
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.DU63ZhEJ9o/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data

jhermsmeier, 2018-05-02:
@andersab looks like you're experiencing #1786, rather than this issue

closed 2018-10-15 by jviotti

Luccasbb2, 2019-06-02:
Vá em fontes de aplicativos, depois Bintray/ stable e desabilita, isso resolveu pra mim.

zahyaa, 2022-06-15:
https://github.com/balena-io/etcher?d_id=58e02706-2ca2-465d-bd23-5b4f5b801e22R#debian-and-ubuntu-based-package-repository-gnulinux-x86x64

Follow this link ti install .deb using termianl commands.