balena-io/etcher #2145: How to build Etcher from source, Void Linux MUSL

kenny-w7, 2018-03-26:
Distro: Void Linux x86_64-MUSL

Hey guys, I'm trying to figure out how to compile Etcher from source so I can have it work on Void Linux.

Makefile:188: No Sentry token found (ANALYTICS_SENTRY_TOKEN is not set)
Makefile:194: No Mixpanel token found (ANALYTICS_MIXPANEL_TOKEN is not set)
Available targets: help info lint lint-js lint-sass lint-cpp lint-html lint-spell test-spectron test-gui test-sdk test-cli test sanity-checks clean distclean changelog webpack package-electron package-cli cli-develop installers-all publish-all electron-develop electron-installer-appimage electron-installer-debian electron-installer-redhat cli-installer-tar-gz publish-aws-s3

jhermsmeier, 2018-03-26:
Have you checked the instructions in the docs, or more specifically [CONTRIBUTING.md#cloning-the-project](https://github.com/resin-io/etcher/blob/master/docs/CONTRIBUTING.md#cloning-the-project)?

lurch, 2018-03-26:
Does Electron even work on Void Linux MUSL? (IIRC MUSL is an alternative libc package?)

kenny-w7, 2018-03-26:
Oh dang that's right, my bad! I forgot about that. 
I will find out soon, lurch.

kenny-w7, 2018-03-26:
https://pastebin.com/VywpfBnR

kenny-w7, 2018-03-26:
npm start

> etcher@1.3.1 start /home/kzn/Compiles/etcher
> electron lib/start.js

events.js:165
      throw er; // Unhandled 'error' event
      ^

Error: spawn /home/kzn/Compiles/etcher/node_modules/electron/dist/electron ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:201:19)
    at onErrorNT (internal/child_process.js:379:16)
    at process._tickCallback (internal/process/next_tick.js:114:19)
    at Function.Module.runMain (module.js:692:11)
    at startup (bootstrap_node.js:194:16)
    at bootstrap_node.js:666:3
Emitted 'error' event at:
    at Process.ChildProcess._handle.onexit (internal/child_process.js:207:12)
    at onErrorNT (internal/child_process.js:379:16)
    [... lines matching original stack trace ...]
    at bootstrap_node.js:666:3
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! etcher@1.3.1 start: `electron lib/start.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the etcher@1.3.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2018-03-26T18_05_49_543Z-debug.log

kenny-w7, 2018-03-26:
Hmmmm... It would appear I'm unable to get Etcher to build on Void MUSL... Not sure why exactly, but the info is there if any of you can better interpret the errors, this is new territory for me unfortunately.

lurch, 2018-03-27:
I've never used Void Linux myself, but have you seen #2011 ?

kenny-w7, 2018-03-27:
./etcher-1.3.1-x86_64.AppImage: No such file or directory
-_- what is wrong with this thing?? Even when I click on it in the file it does the same thing... ugh...
I know flatpak works, do you guys have a flatpak image perhaps?

lurch, 2018-03-28:
> ./etcher-1.3.1-x86_64.AppImage: No such file or directory

IIRC people get that (confusing) error message when trying to run the 64-bit AppImage on a 32-bit Linux distro. But perhaps using MUSL versus glibc also triggers the same error? \*shrug\*

> I know flatpak works, do you guys have a flatpak image perhaps?

#2019

kenny-w7, 2018-03-30:
I can't make AppImages work no matter what I do, a flatpak image would be nice, BUT....
Being able to build-from-source is always more preferable. Its a shame electron isn't supported by Void MUSL. I have to get Electron installed before I can compile Etcher... Haven't been able to get it to install.

kenny-w7, 2018-03-30:
I'm trying to figure out how to get Electron installed on Void MUSL but it seems impossible, I don't know.... :/

kenny-w7, 2018-03-30:
This is what I get when I try to start electron with npm start:
https://pastebin.com/syAbXR90

electron-quick-start, npm start:
https://pastebin.com/PZyba3su

And when trying to launch Etcher:
https://pastebin.com/eLbyrM9v

Jakecheslekov, 2018-03-30:
Can confirm AppImages don't work for me either, says the file doesn't exist even though its been made executable, even when clicking it in the file manager it claims it doesn't exist.

lurch, 2018-04-03:
If even Electron doesn't run on Void MUSL, then there's no chance that Etcher (which is built on top of Electron) will run on Void MUSL.

closed 2018-10-15 by jviotti