balena-io/etcher #2185: PGP issue or documentation update?

ShaunID, 2018-04-06:
- **Etcher version:** 
**N/A**

- **Operating system and architecture:** 
sanderson@neon1604:/media/sanderson/DataDrive/backupforKDE$ uname -a
Linux neon1604 4.13.0-38-generic #43~16.04.1-Ubuntu SMP Wed Mar 14 17:48:43 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
sanderson@neon1604:/media/sanderson/DataDrive/backupforKDE$ lsb_release -a
No LSB modules are available.
Distributor ID: neon
Description:    KDE neon User Edition 5.12
Release:        16.04
Codename:       xenial

- **Image flashed:** 
**N/A**

- **Do you see any meaningful error information in the DevTools?** 
N/A

I was attempting to add Etcher via an Ansible script to rebuild my laptop.  When trying to add the pgp key, it failed.  I tried to add it manually and I am getting the same error. 

```
TASK [apt_repository] **********************************************************
changed: [localhost]

TASK [Add key to install Etcher from repository] *******************************
fatal: [localhost]: FAILED! => {"changed": false, "cmd": "apt-key adv --keyserver pgp.mit.edu --recv 379CE192D401AB61", "failed": true, "msg": "gpg: requesting key D401AB61 from hkp server pgp.mit.edu\ngpg: no valid OpenPGP data found.\ngpg: Total number processed: 0\ngpg: keyserver communications error: keyserver helper general error\ngpg: keyserver communications error: unknown pubkey algorithm\ngpg: keyserver receive failed: unknown pubkey algorithm", "rc": 2, "stderr": "gpg: requesting key D401AB61 from hkp server pgp.mit.edu\ngpg: no valid OpenPGP data found.\ngpg: Total number processed: 0\ngpg: keyserver communications error: keyserver helper general error\ngpg: keyserver communications error: unknown pubkey algorithm\ngpg: keyserver receive failed: unknown pubkey algorithm\n", "stdout": "Executing: /tmp/tmp.SNkkigUmNN/gpg.1.sh --keyserver\npgp.mit.edu\n--recv\n379CE192D401AB61\ngpgkeys: key 379CE192D401AB61 can't be retrieved\n", "stdout_lines": ["Executing: /tmp/tmp.SNkkigUmNN/gpg.1.sh --keyserver", "pgp.mit.edu", "--recv", "379CE192D401AB61", "gpgkeys: key 379CE192D401AB61 can't be retrieved"]}

PLAY RECAP *********************************************************************
localhost                  : ok=22   changed=6    unreachable=0    failed=1   

sanderson@neon1604:/media/sanderson/DataDrive/backupforKDE$ ls -l /etc/apt/sources.list.d/
total 24
-rw-r--r-- 1 root root   57 Apr  6 11:45 dl_bintray_com_resin_io_debian.list
-rw-r--r-- 1 root root   69 Apr  6 11:45 download_virtualbox_org_virtualbox_debian.list
-rw-r--r-- 1 root root   57 Apr  6 11:47 etcher.list
-rw-r--r-- 1 root root  192 Apr  6 11:45 google-chrome.list
-rw-r--r-- 1 root root  102 Apr  6 11:45 neon.list
-rw-r--r-- 1 root root 1525 Apr  6 11:45 teamviewer.list

sanderson@neon1604:/media/sanderson/DataDrive/backupforKDE$ sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/tmp.rPfNUXD8Ug/gpg.1.sh --keyserver
hkp://pgp.mit.edu:80
--recv-keys
379CE192D401AB61
gpg: requesting key D401AB61 from hkp server pgp.mit.edu
gpgkeys: key 379CE192D401AB61 can't be retrieved
gpg: no valid OpenPGP data found.
gpg: Total number processed: 0
gpg: keyserver communications error: keyserver helper general error
gpg: keyserver communications error: unknown pubkey algorithm
gpg: keyserver receive failed: unknown pubkey algorithm
```

I ended up downloading the appImage so this is FYI.

Thanks!!

jhermsmeier, 2018-04-06:
Duplicate of #1786

jhermsmeier, 2018-04-06:
This is an issue with the MIT key server, which has become a bit unstable at times (we've already had 3 or more issues filed on this) :)

jhermsmeier, 2018-04-06:
Closing as duplicate

closed 2018-04-06 by jhermsmeier

ShaunID, 2018-04-06:
I'll use alternate automation methods if the mit server is unreliable.  THANKS!!