balena-io/etcher #2193: it looks like (mac high sierra)

kv71, 2018-04-11:
- **Etcher version:** 
version v1.3.1
- **Operating system and architecture:** 
MAc OS high Sierra version 10.13.4
- **Image flashed:** 
windows 10 , windows server 2016, ect 
- **Do you see any meaningful error information in the DevTools?** 

before i could easy (in high sierra use etcher to burn windows ) but now  it always mentions "It Looks like this is not a bootable image
[image]

lurch, 2018-04-11:
See https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#why-is-my-drive-not-bootable

kenny-w1, 2018-04-13:
You guys should add Solaris, NetBSD & OpenBSD to that list along with Windows 10... FreeBSD will work tho.
I always thought Etcher worked with windows/etc, didn't know it was just like DD...

lurch, 2018-04-16:
See also #210

jhermsmeier, 2018-06-05:
Closing this, as it's documented, and alternatives available

closed 2018-06-05 by jhermsmeier