balena-io/etcher #2194: Etcher 1.3.1 detected potential corruption issues during validation

Webbanner, 2018-04-12:
[image]
Happened during validating SAMSUNG micro SD card EVO, 16gb

Webbanner, 2018-04-12:
But successfully validating KINGSTON 8GB micro SD card, class 4

lurch, 2018-04-12:
Is it possible that your "SAMSUNG UHS class 1" is broken / flaky / faulty?

Sometimes some SD card-readers aren't fully compatible with certain types of SD cards, and it's possible to avoid this problem by using a different SD card-reader.

jhermsmeier, 2018-04-12:
No, there shouldn't be any issues regarding the SD card classes; any should work.
This is most likely to be an issue with the hardware, or sporadic write errors – i.e. the SD card itself, or the adapter – you can retry it (if you haven't already), and if it consistently fails, then there's something wrong with the hardware.

Webbanner, 2018-04-13:
Might be the SAMSUNG sd card broken...use same card reader on KINGSTON..it works like charm..

rowanbradley, 2018-04-13:
I've got the same issue. I am trying to use a 32GB micro-SD card, class 10, marked HC (no other brand marking). I have tried more than once, and I have bought another 32GB card, and it still does it. I really really don't think this is a hardware problem, or a "sporadic write errors" problem. Can anyone suggest how to move forward, i.e. how to overcome this problem, or how to investigate further what is wrong>?

Thanks - Rowan

lurch, 2018-04-14:
You could use h2testw to check that your SD card is behaving correctly, and not silently corrupting data.

tordf, 2018-04-18:
Have the same issue. Tried with both a Sandisk Utra 16 GB and a Panasonic 8GB. Both are class 10 HC 1. The Panasonic card is brand new.