balena-io/etcher #2197: Raspberry Pi Compute Module (and rpi0) enumeration fails

curcuz, 2018-04-12:
- **Etcher version:**  1.4.1
- **Operating system and architecture:**  OSX 10.13.4
- **Image flashed:**  resinOS v2.12.5-rev2 prod
- **Do you see any meaningful error information in the DevTools?**  yes, see below:
[image]

```
gui.js:105102 {"name":"Error","message":"ENOENT: no such file or directory, open 'lib/blobs/usbboot/raspberrypi/bootcode.bin'","cause":{"errno":-2,"code":"ENOENT","syscall":"open","path":"lib/blobs/usbboot/raspberrypi/bootcode.bin"},"isOperational":true,"stack":"Error: ENOENT: no such file or directory, open 'lib/blobs/usbboot/raspberrypi/bootcode.bin'","errno":-2,"code":"ENOENT","syscall":"open","path":"lib/blobs/usbboot/raspberrypi/bootcode.bin"}
console.(anonymous function) @ gui.js:105102
gui.js:105102 etcher:sdk:usbboot Serial number index: 0 +1m
gui.js:105102 etcher:sdk:usbboot [20:18] -> 10% +0ms
gui.js:105102 etcher:sdk:usbboot Opening device: 20:18 +0ms
gui.js:105102 etcher:sdk:usbboot Claiming interface: 0 +0ms
gui.js:105102 Unhandled rejection Error: ENOENT: no such file or directory, open 'lib/blobs/usbboot/raspberrypi/bootcode.bin'
```

lurch, 2018-04-12:
Just FYI: even without this error, you'd still be unable to enumerate the CM1 or RPi Zero on OSX, due to a bug in the bootROM in the BCM2835 https://www.raspberrypi.org/documentation/hardware/computemodule/cm-emmc-flashing.md

curcuz, 2018-04-12:
@lurch thx for the info - FYI this also happens on CM3

Shou, 2018-04-12:
This doesn't seem to affect Linux btw, usbboot runs as usual here.