balena-io/etcher #2201: What technology is use to create Etcher?

stephane-archer, 2018-04-16:
Etcher is beautiful cross platform app.
I would like to know what technology and library did you use?

Thank you 😄

lurch, 2018-04-16:
I think https://github.com/resin-io/etcher/blob/master/docs/ARCHITECTURE.md should answer your question? :slightly_smiling_face: 
(although I think some small parts of that might be out-of-date now)

Shou, 2018-04-16:
Yup, what @lurch said – we should probably update that document. We are phasing out the use of Angular in favour of React. Bootstrap and Sass are also mostly being replaced by an 'in-house' library called [rendition](https://github.com/resin-io-modules/rendition) that is React with styled-system and styled-components. We also use [Lodash](https://lodash.com) quite extensively.