balena-io/etcher #2208: Validating is very slow

snowdream, 2018-04-17:
- **版本 1.3.1 (1.3.1)** 
- **macbook pro** 
- **Validating is very slow:** 
image flashed, then validating is very very slow.

[image]

jhermsmeier, 2018-04-23:
How fast was the flashing compared to this?

fjeddy, 2019-02-18:
Same issue here. Validating takes 5-6 times longer then the actual flashing. Macbook Pro.

mrmuminov, 2022-10-23:
I have this problem too
[image]
I'm waiting now...

>> Flashing is 6-10 MBps

zapta, 2025-01-06:
Same here. Windows 10. Validation is slower by at least one order of magnitude than flashing.