balena-io/etcher #2210: Prevent "Format disk?" popup on Windows

jhermsmeier, 2018-04-17:
I stumbled upon the reason & the solution for the annoying "Format disk?" popup on Windows, when a storage device without a recognized partition table is detected by the system:

>  [...] when a new volume created in system, it is possible that the system bring up a windows prompt for “Format disk?”, that is not what I expecting, I need to format the volume using my special parameter. To disable this prompt, need to stop/restart server “ShellHWDetection“, there is [a sample about how to control a service](http://msdn.microsoft.com/en-us/library/windows/desktop/bb540474%28v=vs.85%29.aspx) [...].

Source: [How to format a RAW disk in windows by C++](https://wreckorwalker.wordpress.com/2014/06/30/how-to-format-a-raw-disk-in-windows-by-c/)

I suppose we could implement the service control, to silence the shell's hardware detection while Etcher is processing devices.

lurch, 2018-04-18:
Presumably you'd need to be extra-careful that the `ShellHWDetection` service gets started again when Etcher exits and also if it crashes?

refthoom, 2018-10-23:
This pop up seems to interfer with validating on Win10. When I etch a micro SD card progress goes to 100% without issue but after validation starts, it hangs either from the start or after 2 - 7 % and then ends with an error that etcher is unable to write to the drive. I have noticed two things in regard to the "format..." popup. One is that if etcher's validating does not start at all, the pop up stays away until after I click away etcher's error. If etcher is able to start validating, it hangs a soon as the pop up shows, wgch some times takes from 5 - 15 seconds. 

I have witnessed this behavior on two different desktops and a laptop, all running Win10 that I keep updated regularly (but not too regularly to prevent new bugs from M$). To exclude the SD card I have used two 2GB and 1 4GB card with the exact same results. Also, witnessed the same behaviour for v1.4.4 and 1.4.5. Running as adminstrator makes no difference. Different card readers, with or without SDcard-to-microSdcard adapters also makes no difference.

So IMHO it is either Win10 or Etcher :-)

BTW If the cards were fine after this I wouldn't give it a second thought, however I do run into issues. As I use these SD cards for my raspberry pi's who have been running without issues for a while, ssh'ing into them won't work. Running them with display and keyboard is ok, I can ping them, just not SSH. And yes I have checked all and tried various settings, including the most obvious enabling SSH from the raspbian config utility. 

Apart from all this, I do not understand the error. When validating etcher shouldn't need to write to the disk, just read, right?

Anyway, maybe it's all just a weird coincidence, don't know. Just thought I'dmention it here.

drakorg, 2019-05-18:
Same issue here.
It's kind of tedious to have windows popup 100 times the "format disk?" message, while etcher is still in validation phase. Whats going on?

knyzorg, 2020-06-11:
Honestly, I'm more annoyed at how it keeps popping up afterwards. I have the USB plugged in so I don't lose. Windows begging to blow my gparted drive away is driving me nuts.

lurch, 2020-06-12:
Related to #3197 ?

loophole64, 2021-03-02:
Any resolution to this? I am currently having this problem. Writing a 64 gb usb drive and during validation I get the dialog prompting me to format each partition windows thinks it recognizes. I didn't have this behavior on another machine. Both machines were windows 10. I'm writing a bootable linux partition for a Jetson Nano.

sehrguey, 2022-05-20:
the issue was around for at least 10 years, nether removing usb disk letter with diskmgmt.msc nor giving the disk partition id=64 with diskpart.msc ever worked for me. seems there's still no reliable solution for the MS crap

Lightnet, 2023-01-05:
Did some searching. It the AutoPlay settings for windows 10. It can be disable. Open the search bar type AutoPlay. You can turn it off.

Use AutoPlay for all media and devices.