balena-io/etcher #2224: Add Micro:bit support to the SDK

jviotti, 2018-04-22:
See http://microbit.org/

The existing module to flash this board is https://github.com/Microsoft/dapjs. There is also https://github.com/ntoll/uflash, which seems to be only for flashing Python code to the micro:bit.

We need to flash using deltas, to reduce the time needed to flash the whole board’s memory, which currently takes about 15 seconds. The approach I planned to develop is to divide the memory in blocks, hash them, and also have superhashes based on the block’s hashes. This means we can only send the blocks that are needed.

For scanning, the board exposes itself through USB as:

```
DAPLink CMSIS-DAP:

  Product ID: 0x0204
  Vendor ID: 0x0d28
  Version: 10.00
  Serial Number: 9900000031864e45001d1007000000680000000097969901
  Speed: Up to 12 Mb/sec
  Manufacturer: ARM
  Location ID: 0x14100000 / 6
  Current Available (mA): 500
  Current Required (mA): 500
  Extra Operating Current (mA): 0
  Media:
    VFS:
      Capacity: 8.5 MB (8,454,144 bytes)
      Removable Media: Yes
      BSD Name: disk2
      Logical Unit: 0
      Partition Map Type: Unknown
      USB Interface: 0
```

The best way to detect a micro:bit is to check that the serial number starts with “9900”.

lurch, 2018-04-22:
> The existing module to flash this board is https://github.com/Microsoft/dapjs

The README.md in that repo says "Note: this package is now maintained by ARM at https://github.com/ARMmbed/dapjs " ??

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti