balena-io/etcher #2226: Doesn't do anything on MacOS

gordonmleigh, 2018-04-23:
- **Etcher version:** 1.4.3
- **Operating system and architecture:** MacOS 10.13.3 x64
- **Image flashed:** elementaryOS 0.4.1

Repro:

1. Select image
2. Select drive (onboard SD card reader drive on MacBook Pro)
3. Click "Flash!"
4. The button briefly says "Starting" then goes back to "Flash!" - nothing else happens
5. Full logs below:

```
Mon Apr 23 2018 09:07:20 GMT+0100 (BST) Application start ({"packageType":"dmg","version":"1.4.3"})
gui.js:105081 Mon Apr 23 2018 09:07:20 GMT+0100 (BST) Checking for updates ({"currentVersion":"1.4.3","stable":true,"updateSemverRange":"<2.0.0","includeUnstableChannel":false})
gui.js:105081 etcher:sdk:scanner start +116ms true
gui.js:105081 Mon Apr 23 2018 09:07:21 GMT+0100 (BST) [object Event]
gui.js:105081 Mon Apr 23 2018 09:07:22 GMT+0100 (BST) Update notification skipped ({"reason":"Latest version"})
gui.js:105081 Mon Apr 23 2018 09:07:43 GMT+0100 (BST) Open image selector
gui.js:105081 Mon Apr 23 2018 09:07:47 GMT+0100 (BST) Select image ({"path":"/Users/gordonmleigh/Downloads/elementaryos-0.4.1-stable.20180214.iso","extension":"iso","size":{"original":1498415104,"final":{"estimation":false,"value":1498415104}},"hasMBR":true,"hasGPT":true,"partitions":[{"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","id":"665E2759-F498-4972-A6C7-13DF68D09285","name":"卉䡏批楲⁤卉O卉䡏批楲d灁汰","firstLBA":0,"lastLBA":2924652,"extended":false},{"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","id":"B57B8352-6116-4D37-8A04-56DA891DB2F5","name":"卉䡏批楲d灁汰e灁汰","firstLBA":5364,"lastLBA":10035,"extended":false}],"logo":false,"bmap":false})
gui.js:105081 Mon Apr 23 2018 09:07:48 GMT+0100 (BST) Open modal ({"name":"drive-selector"})
gui.js:105081 Mon Apr 23 2018 09:07:49 GMT+0100 (BST) Toggle drive ({"drive":{"size":31914983424,"isVirtual":false,"enumerator":"diskutil","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"adapter":"blockdevice","isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"mountpoints":[{"path":"/Volumes/USBOOT","label":"USBOOT"}],"description":"APPLE SD Card Reader Media","isUAS":null},"previouslySelected":false})
gui.js:105081 Mon Apr 23 2018 09:07:50 GMT+0100 (BST) Modal accepted ({"name":"drive-selector"})
gui.js:105081 etcher:sdk:scanner stop +30s true
gui.js:105081 Mon Apr 23 2018 09:07:50 GMT+0100 (BST) Flashing 0 devices, 0% at 0 MB/s (total 0 MB/s) eta in undefineds with 0 failed devices
gui.js:105081 Mon Apr 23 2018 09:07:50 GMT+0100 (BST) Flash ({"image":"/Users/gordonmleigh/Downloads/elementaryos-0.4.1-stable.20180214.iso","drives":["/dev/disk2"],"uuid":"b2c8925a-cf6b-466a-a746-bfb7bd49c238","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
gui.js:105081 Flash results Object
gui.js:105081 Terminating IPC server
gui.js:105081 Mon Apr 23 2018 09:07:51 GMT+0100 (BST) Elevation cancelled ({"image":"/Users/gordonmleigh/Downloads/elementaryos-0.4.1-stable.20180214.iso","drives":["/dev/disk2"],"uuid":"b2c8925a-cf6b-466a-a746-bfb7bd49c238","unmountOnSuccess":true,"validateWriteOnSuccess":true})
```

jhermsmeier, 2018-04-23:
@gordonmleigh did you see the dialog that prompts you for your credentials? The logs show it was canceled, and hence the flashing didn't start.

gordonmleigh, 2018-04-23:
Didn't see anything come up, but as you can see it says it was cancelled 1s after it started, so it's not like it timed out.  Also see where it says "Flashing 0 devices", I assume that's wrong?

jhermsmeier, 2018-04-23:
Yeah, there seems to be something odd going on. Just wondering how we might reproduce this. Was this a one-time thing, or does it behave like this persistently for you?

gordonmleigh, 2018-04-23:
I tried it 3 or 4 times, same result each time.

jhermsmeier, 2018-04-23:
Does it behave differently with a different image / sd-card?

closed 2018-10-15 by jviotti