balena-io/etcher #2235: Get some sort of unique identifier for USB ports

jviotti, 2018-04-23:
This should probably be implemented in `drivelist`, and will be used to then determine which USB port is which (to integrate them with the LEDs)

jviotti, 2018-04-23:
Looks like we might have solved this long ago when working on the "headless Etcher" project long ago. Worth checking there.

jviotti, 2018-04-23:
From @petrosagg :

> each port has a unique path as it traverses the various hubs and ports. We can find the answer in /dev/disk/by-path/*

jviotti, 2018-04-23:
We do need to do this cross-platform though.

lurch, 2018-04-23:
If you're still able to find it, IIRC I once made a thread about this in Flowdock (I managed to make some persistent symlinks in `/dev/` by adding a new config file in `/etc/udev/rules.d/`). But again, that was just on Linux, I didn't do similar investigation on Windows or Mac.

zvin, 2018-05-03:
https://www.flowdock.com/app/rulemotion/r-etcherprv/threads/8f-2QPV-bcT3LEb-WQv6F_tqDsV

zvin, 2018-05-03:
also https://www.flowdock.com/app/rulemotion/r-etcherprv/threads/uDXBQzzC3QX7mez9am3XIZ461_r

zvin, 2018-05-03:
There is 2 problems here:
 - identify each usb port uniquely
 - know to which of these ports the mass storage device is connected

zvin, 2018-05-03:
For identifying usb ports with node-usb, we can probably use `[ device.busNumber ].concat(device.portNumbers)`.
I'm not sure the `busNumber` is absolutely needed.

zvin, 2018-05-03:
On Linux we do know the port a mass storage is connected to by the name of the link in `/dev/disk/by-path`.

jhermsmeier, 2018-05-09:
This has been implemented & merged in https://github.com/resin-io-modules/drivelist/pull/278 – but the new drivelist version isn't published yet, due to some minor issues with the prebuilds to be sorted out.

lurch, 2018-05-14:
The referenced drivelist PR only adds this feature for Linux, but earlier in this issue @jviotti said "We do need to do this cross-platform though."
Is it correct for this issue to be closed?

jhermsmeier, 2018-05-14:
@lurch support for this on other platforms will be added eventually (as we want to maintain feature parity across platforms in drivelist), but this issue is Etcher Pro related

commit f5fd2f2b (2018-05-10): Merge pull request #2325 from resin-io/update-drivelist

upgrade(package): Bump drivelist 6.1.7 -> 6.2.2
Close #2235