balena-io/etcher #2239: Add an image selection dropdown to pick from selecting a file or duplicating a drive

jviotti, 2018-04-23:
[image]

jviotti, 2018-04-23:
I just noticed we need to sort out what to present if, in the case of Etcher Pro, the source drive has multiple partitions. Should we make the user pick? cc @konmouz

jviotti, 2018-04-23:
This is blocked by https://github.com/resin-io/etcher/issues/2230

jviotti, 2018-04-26:
When selecting one of these options, there should be a label that says "Remove" that makes the user go back to the initial screen.

jviotti, 2018-04-26:
If the user selects a drive to clone, then the UI shouldn't allow the user to select that same drive as the target.

konmouz, 2019-02-26:
@jviotti, We should add image streaming as a third option.

[image]

konmouz, 2019-06-03:
New screen: https://zpl.io/bAvWJMq

closed 2020-10-02 by thundron