balena-io/etcher #2240: Implement multi-writes success/error screens

jviotti, 2018-04-23:
We want to keep the banner, so we must figure out a way to display the banner while showing meaningful summaries.

The logic is the following:

- If at least one drive was successful, then the user goes to the success screen
- If all drives fail, you get the usual error modal with the "Retry" action

**Let's make sure this is touch friendly**

[image]

jhermsmeier, 2018-05-14:
ping @konmouz ^

konmouz, 2018-05-16:
@Shou, what do you think?

[image]

konmouz, 2018-05-16:
maybe the CTA can be 'Re-flash failed targets'

lurch, 2018-05-18:
Something that's just occurred to me - when the "Failed drives" modal is being displayed on the EtcherPro, perhaps it'd be more useful to display drive-paths like "USB slot 3" rather than "/dev/sdf" ?

dazz100, 2018-07-26:
Hi
I am trying to use several 4-slot SD card readers.  If a SD card fails, Etcher doesn't tell me which slot the failed card is in.
Based on practical experience, I think the best solution is to give the user all of the information available to Etcher about the slot.  That would include both  /dev/sdf    usb 01:slot 02 etc.

konmouz, 2019-06-03:
EP success screen - no banner screen - @konmouz

konmouz, 2019-06-24:
[image]
[image]
[image]

https://zpl.io/ankLE6J -- @thundron new screens

dazz100, 2019-06-24:
Hi 
Received copies of the screen shot by email without comment. 
I like the screenshot layouts but
I am using one of these :  [USB3.0 to 4xSD Card Reader](http://www.chinagallop.com/products_detail/productId=752.html).  
I don't know what device you are testing on so I can't tell from the screenshots if your code will identify the slot of a failed flash within a single multi-card writer/reader.  If it can, that would be great.
If it can't, then this code is doesn't help anyone with a multi-card reader/writer.

I am more than happy to do testing if that will help.

zvin, 2019-06-26:
@dazz100 there's no way to identify a port number on a multi-card reader.
We may have a physical path (like `/dev/disk/by-path/pci-0000:00:14.0-usb-0:3:1.0-scsi-0:0:0:0` or `/sys/devices/platform/soc/3f980000.usb/usb1/1-1/1-1.5/1-1.5:1.0` on Linux for example) but that won't help you identifying the physical port.

closed 2021-08-17 by thundron