balena-io/etcher #2245: Add a cancel button the UI

jviotti, 2018-04-23:
This will allow the user to cancel all flashes (for now), and potentially particular ones in the future.

jviotti, 2018-04-23:
This is blocked by https://github.com/resin-io/etcher/issues/2234

jviotti, 2018-04-23:
We also need some designs from @konmouz

lurch, 2018-04-23:
Duplicate of #1791 ?

jviotti, 2018-04-26:
[image]

jviotti, 2018-04-26:
We should also add a modal that says "Are you sure you want to cancel?" or something like that

closed 2018-05-03 by Shou

konmouz, 2019-06-26:
[image]

Updated design - https://zpl.io/aRzJezg