balena-io/etcher #2250: Use LEDs to mirror state of the Etcher Pro

jviotti, 2018-04-23:
- When inserting the drive, the corresponding LED turns on
- When removing the drive, the corresponding LED turns off
- When validating the drive, the corresponding LED should blink with purple
- When a failure occurs, the corresponding LED should be red
- When the flash is successful, the corresponding LED should be green
- When writing, the corresponding LED should be blinking with orange

lurch, 2018-04-23:
What colour should the LED be when a drive is inserted but before you've started writing to it?

jviotti, 2018-04-24:
Not sure yet. I left unspecified since its probably not worth to raise a discussion for it yet (and be updated really easily). We can use any color not used by the other states, like blue.

lurch, 2018-04-24:
I wonder if it'd be worth updating parts of the GUI, so that the colours used in the GUI match the colours used on the LEDs? (which might make it much more obvious what the different LED colours mean?)

curcuz, 2018-05-21:
@jhermsmeier please issue a WIP PR for this

konmouz, 2019-06-03:
- Write the library to control the LEDs
- Define the LEDs behaviour according to EP function

konmouz, 2020-03-12:
Prompt to plug source: 1st LED > blue - breathing
Source plugged: 1st LED > blue - on
Prompt to plug target: 2nd to 16th LED > white - breathing (or all LEDs in case of an external source)
Target plugged: LED of the plugged drive > white - on
Flashing > green - blinking
Validating > purple - blinking
Successfully flashed > green - on
Failed drives > red - on

lurch, 2020-03-12:
Relates to #3046 ?

konmouz, 2020-05-18:
Diagram: https://docs.google.com/drawings/d/12n5z5qhw6eriHyaDEnId2Jhgjt95wBn4PXR1pkqCORQ/edit?usp=sharing

[image]

closed 2021-08-17 by thundron