balena-io/etcher #2251: Write a container to handle Wifi configuration

jviotti, 2018-04-23:
**This code should live in a separate `etcher-pro` repository along with the resin specific code. **

The idea is that the Etcher Pro application will consists of two containers: the Etcher container, and the Wifi container. The wifi container will contain everything needed to communicate with NetworkManager through its DBUS API, and will expose an HTTP API so we can communicate to it.

There is a very good Python module to do this (`python-networkmanager`). We are unsure of Node.js alternatives, so we can either try some from NPM, or just write this contain in Python.

lurch, 2018-04-23:
How does this relate to https://github.com/resin-io/resin-wifi-connect ? Or is that something entirely different?

jviotti, 2018-04-30:
@jhermsmeier Turns out we can use wifi connect in the end, with its recent modifications. See https://github.com/resin-io-projects/resin-wifi-connect-api as an example.

Shou, 2018-04-30:
@jviotti are we allowing people to set it with other devices _and_ the UI now then?

closed 2021-08-17 by thundron