balena-io/etcher #2252: Implement a UI to interact with the Wifi container

jviotti, 2018-04-23:
Obviously depends on https://github.com/resin-io/etcher/issues/2251.

This will be a separate wifi configuration panel, probably accessible through a wifi icon on the top right corner of the app (like the settings cog). This is Etcher Pro specific, so it should not be enabled for desktop Etcher.

jviotti, 2018-04-23:
We're pending some designs for this. @konmouz

konmouz, 2018-05-02:
@jviotti @Shou let me know what you think of these screens:

[image]
[image]
[image]

jviotti, 2018-05-03:
@konmouz What does "Advanced Configuration" show? Overall looks good to me. @majorz Are all the features in those screens supported by wifi connect?

konmouz, 2018-05-03:
@jviotti, this is just in case we want to include further or advanced options. We can remove it if this is irrelevant.

curcuz, 2018-05-09:
@jviotti @konmouz I guess advanced stuff would be static IP config, or anything non DHCP standard

closed 2021-08-17 by thundron