balena-io/etcher #2266: Implement a new success banner for Etcher Pro

jviotti, 2018-04-24:
This banner should not contain external links, since the user won't be able to follow it. It can be used for promotional purposes, or to perform an action from within the banner.

lurch, 2018-04-24:
Connects to https://github.com/resin-io/etcher-homepage

lurch, 2018-04-25:
And it's just occurred to me that having an Etcher-Pro-specific success banner also also makes sense from a marketing point of view, because you might want to continue to promote Etcher Pro on the regular Etcher banner, but there's no point advertising Etcher Pro on the success banner that would appear on an Etcher Pro itself :wink: (IYSWIM)

konmouz, 2019-06-03:
Prepare screens - @konmouz

jviotti, 2022-08-15:
Closing stale issues

closed 2022-08-15 by jviotti