balena-io/etcher #2267: [Bug Report] Etcher 1.4.3 - Drive Select GUI Error

Richyread, 2018-04-24:
- **Etcher version:** 1.4.3
- **Operating system and architecture:** Win 10(64)
- **Image flashed:** Raspbian Stretch Lite (13-03-2018)
- **Do you see any meaningful error information in the DevTools?** See screenshot attached.

I have upgraded my Etcher installation this morning and encountered an issue with writing an image to an SD Card - specifically that the device/drive selection step appears to accept and then immediately cancel the choice within the GUI ~ see shot below for better description 😄 

[image]

**Workflow prior to upgrade:**

- Insert SD Card into reader

- Close Win10's 'Please Format Drive' prompts

- Launcher Etcher

- Select image file (including .ZIP archived images)

- Select USB Card Reader

- Click 'Flash'

Etcher would then in chain of actions, Format the Drive(s), Extract the image, Write the image, Tidy up, Eject SD Reader.

Since the upgrade I'm unable to proceed past the 'Select Drive' stage.

Screenshot of Etcher_Console, below.

[image]

Hope this helps - let me know if you need anything further etc.

Many thanks,

Richy.

PecoraInPannaCotta, 2018-04-24:
Same on ubuntu 17.10

nex86, 2018-04-24:
same issue. going back to 1.3.1!

Shou, 2018-04-24:
What happens when you double-click?

nex86, 2018-04-24:
it keeps deselecting even when you double click.

PecoraInPannaCotta, 2018-04-24:
If you go faster than the GUI, you can actually flash the ISO, so is confirmed is just a UI problem

fivestones, 2018-04-24:
I wish I were as fast as you :-)

Shou, 2018-04-24:
I've identified the issue and fixed it in the above PR. We'll have 1.4.4 out shortly.

closed 2018-04-25 by jhermsmeier