balena-io/etcher #2271: Error Launching Etcher, if Etcher's Internet Requests are Blocked by Firewall

ClosetCoder, 2018-04-24:
- **Etcher version:** 
     v1.4.3

- **Operating system and architecture:** 
     mac OS High Sierra v10.13.4 (17E199) - i7 Quad 2.4GHz - 16GB RAM - 512GB SSD
- **Image flashed:** 
     None - Error occurs simply opening app if internet access is blocked
- **Do you see any meaningful error information in the DevTools?** 
     Yes:  [Error in Console](https://ibb.co/fBXJ1x)
- **Error message produced by Etcher**
     Yes:  [Error Message](https://ibb.co/cRGd1x) w/'OK' to proceed
- **Clicking 'OK' on message allows Etcher to run**
     Yes
- **Define Blocked in Issue Title**
     Rules are added to firewall to block ALL internet requests made by Etcher

ClosetCoder, 2018-04-25:
Reviewing the errors produced in the console, ...yes, it appears to be the same other than the version number of Etcher. I considered this before submitting, but since Etcher had jumped up so many iterations from v1.3.1 to the now v1.4.3, ...I felt inclined to resubmit it. Not being part of the dev team, ...I wanted to keep it on the forefront of everyone's mind.

lurch, 2018-04-25:
@jhermsmeier this should probably be closed as a duplicate.

jhermsmeier, 2018-04-25:
Duplicate of #2127

closed 2018-04-25 by jhermsmeier