balena-io/etcher #2274: Missing bindings file when using windows build of etcher cli

feistus, 2018-04-25:
- **Etcher version:** etcher(cli) 1.4.2
- **Operating system and architecture:**  Win7 64bit
- **Image flashed:** usbimage.img
- **Do you see any meaningful error information in the DevTools?** error output in cli, see below

Starting and flashing with the gui version works, but using the cli in windows gives me this error as soon as i provide the image parameter.

Could not locate the bindings file. Tried:
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\build\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\build\Debug\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\build\Release\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\out\Debug\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\Debug\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\out\Release\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\Release\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\build\default\drivelist.node
 → C:\snapshot\etcher-cli-1.4.3-windows-x64-app\compiled\6.10.3\win32\x64\drivelist.node

AlexanderPinkerton, 2018-07-11:
Certainly still an issue.

SuperJMN, 2018-07-19:
Still not fixed?

jhermsmeier, 2018-07-19:
@SuperJMN there hasn't been a release since, so no, that's still not fixed yet.

FabioMen10, 2018-07-23:
Hoping that the best way to use this amazing tool is that Etcher works well. For me, at the moment, the same problem occurs.
[image]

kelson42, 2018-09-18:
The file is there `node_modules/drivelist/build/Release/drivelist.node`, but looks like not to be at the right place!

kelson42, 2018-09-27:
Offering 200 USD to anybody getting this problem fixed and ticket closed :)

davidmdem, 2018-10-04:
It looks like this or something similar happened in issue #1706. Some problem with relative directories and the way the application is built.

@jviotti [Suggests that you can install via npm](https://github.com/resin-io/etcher/issues/1706#issuecomment-325415169).

kelson42, 2018-10-09:
We definitely face a regression here. The soft had a really similar error in the past, see #1706 and it has been fixed in v1.1.2... Might be helpful to fix that one (see PR https://github.com/resin-io/etcher/pull/1713)!

abdoullahi-malagouen, 2018-10-24:
is someone looking at this issue ?

rgaudin, 2018-11-26:
Still present in `1.4.8`...

thundron, 2018-12-31:
We're looking at it. It's probably due to a recent rework in the CI, it should be fairly trivial and @kelson42 good guess on the PR, thanks!