balena-io/etcher #2278: Does not work with Windows' "Mapped network drives"

jacobq, 2018-04-25:
- **Etcher version:** 1.4.3
- **Operating system and architecture:** Windows 10 Pro x64 (VMware)
- **Image flashed:** [`2018-04-18-raspbian-stretch.img`](https://www.raspberrypi.org/downloads/raspbian/)
- **Do you see any meaningful error information in the DevTools?** Yes, an `ENOENT` error suggesting that node could not open the file. Though I was able to browse to it. Note that it works when the path is specified as `\\my-server\share\path\to\file` instead of `X:\path\to\file` with `X:` mapped to `\\my-server\share`

```
Wed Apr 25 2018 15:58:37 GMT-0500 (Central Daylight Time) Flashing 0 devices, 0% at 0 MB/s (total 0 MB/s) eta in undefineds with 0 failed devices
gui.js:105081 Wed Apr 25 2018 15:58:37 GMT-0500 (Central Daylight Time) Flash ({"image":"S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img","drives":["\\\\.\\PhysicalDrive1"],"uuid":"c0a7f870-9be4-4937-baf6-091b9377241c","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Elevating command: C:\Users\jacob\AppData\Local\Programs\etcher\Etcher.exe C:\Users\jacob\AppData\Local\Programs\etcher\resources\app.asar
gui.js:105081 Successfully connected to IPC server: etcher-server-6772, socket root C:\Users\jacob\AppData\Local\Temp\
gui.js:105081 Image: S:\Engineering\Reference\RPi\2018-04-18-raspbian-stretch.img
gui.js:105081 Devices: \\.\PhysicalDrive1
gui.js:105081 Umount on success: true
gui.js:105081 Validate on success: true
gui.js:105081 Error: ENOENT: no such file or directory, stat 'S:\Engineering\Reference\RPi\2018-04-18-raspbian-stretch.img'
gui.js:105081 Wed Apr 25 2018 15:58:37 GMT-0500 (Central Daylight Time) Flash error ({"error":{"name":"Error","message":"ENOENT: no such file or directory, stat 'S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img'","stack":"Error: ENOENT: no such file or directory, stat 'S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img'","code":"ENOENT","syscall":"stat","errno":-4058},"image":"S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img","drives":["\\\\.\\PhysicalDrive1"],"uuid":"c0a7f870-9be4-4937-baf6-091b9377241c","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Wed Apr 25 2018 15:58:37 GMT-0500 (Central Daylight Time) Open modal ({"name":"warning"})
gui.js:105081 {"stack":"Error: ENOENT: no such file or directory, stat 'S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img'","message":"ENOENT: no such file or directory, stat 'S:\\Engineering\\Reference\\RPi\\2018-04-18-raspbian-stretch.img'","name":"Error","code":"ENOENT","syscall":"stat","errno":-4058}
console.(anonymous function) @ gui.js:105081
logException @ gui.js:114128
exports.report @ gui.js:23732
(anonymous) @ gui.js:146185
processQueue @ gui.js:82229
(anonymous) @ gui.js:82273
$digest @ gui.js:83368
$apply @ gui.js:83666
(anonymous) @ gui.js:85513
completeOutstandingRequest @ gui.js:71660
(anonymous) @ gui.js:71940
wrapped @ gui.js:105442

```

[image]
[image]
[image]

lurch, 2018-04-25:
Looks like a very similar problem to #1436

jacobq, 2018-04-25:
Might also be related to https://github.com/nodejs/node/issues/8823

xyvyx1, 2018-12-08:
I was able to solve this on windows 10 (version 1803) by adding the EnableLinkedConnections registry entry as described here:

https://support.microsoft.com/en-us/help/3035277/mapped-drives-are-not-available-from-an-elevated-prompt-when-uac-is-co

I didn't attempt either method 1 or 2 from the "workaround" section.  Initially, I attempted to change the permissions of the shared folder on my Synology volume to allow all guest/default & domain-user accounts read-only access, but that didn't seem to make a difference.

guythnick, 2019-03-20:
Agreed with @xyvyx1 that the workaround below was the only way to fix the issue with mapped drives on Windows 10. 

```
Detail to configure the EnableLinkedConnections" registry entry

In Registry Editor, locate and then click the following registry subkey: 
HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows\CurrentVersion\Policies\System
Right-click Configuration, click New, and then click DWORD (32-bit) Value.
Name the new registry entry as EnableLinkedConnections.
Double-click the EnableLinkedConnections registry entry.
In the Edit DWORD Value dialog box, type 1 in the Value data field, and then click OK.
Exit Registry Editor, and then restart the computer.
```

thundron, 2019-04-18:
@jacobq Did you already try @xyvyx1 's solution by chance? Are you still having this issue?

jacobq, 2019-05-13:
@thundron I have not taken the time to investigate the problem any further. I just copy the image to a local disk first now.

closed 2019-05-13 by jacobq