balena-io/etcher #2281: Multiple drives labeled as containing source image

jhermsmeier, 2018-04-26:
- **Etcher version:** at least 1.3.1+
- **Operating system and architecture:** Linux
- **Image flashed:** N/A

If the selected image is on `/dev/sdc`:

[image]

If the selected image is on `/dev/sda`:

[image]

```js
[ { enumerator: 'lsblk',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/sda',
    raw: '/dev/sda',
    description: 'Samsung SSD 850',
    error: null,
    size: 120034123776,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [ { path: '/' }, { path: '/boot/efi' } ],
    isReadOnly: false,
    isSystem: true,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null },
  { enumerator: 'lsblk',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/sdb',
    raw: '/dev/sdb',
    description: 'SD Transcend',
    error: null,
    size: 31914983424,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [],
    isReadOnly: false,
    isSystem: false,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null },
  { enumerator: 'lsblk',
    busType: 'UNKNOWN',
    busVersion: '0.0',
    device: '/dev/sdc',
    raw: '/dev/sdc',
    description: 'AS2115',
    error: null,
    size: 1000204886016,
    blockSize: null,
    logicalBlockSize: null,
    mountpoints: [ { path: '/media/jonas/R5-D4' } ],
    isReadOnly: false,
    isSystem: true,
    isVirtual: null,
    isRemovable: null,
    isCard: null,
    isSCSI: null,
    isUSB: null,
    isUAS: null } ]
```

lurch, 2018-04-26:
I wonder if it might be getting confused by the `path: '/'` part? i.e. perhaps it thinks all other mountpoints (such as `'/media/jonas/R5-D4'`) are simply subdirectories of `'/'` and therefore `/dev/sda` always "contains" the image ?  (although I wonder why this bug hasn't popped up before now)

Shou, 2018-04-27:
Wouldn't that also affect SD Transcend?

ethanm124816, 2018-08-25:
I have the same problem, really hope someone finds a solution to this. thx!

jhermsmeier, 2018-08-28:
@ethanm124816 which version of Etcher are you running?

ethanm124816, 2018-08-28:
actually, I've already uninstalled the app so I would not know. but thanks
for the help.

hvfrancesco, 2018-10-17:
same problem here, completely unusable

thundron, 2018-11-07:
@hvfrancesco Etcher version number?

zvin, 2018-12-11:
I'm not sure we need to fix anything about this issue, except maybe changing the label from "Drive Contains Image" to "The image path starts with one of this drive mountpoints".

It would not make sense to allow flashing `/dev/sda` mounted at `/` if the image is in `/dev/sdc` mounted on `/media/jonas/R5-D4`. (As soon as you unmount `/dev/sda`, the image will disappear).

lurch, 2018-12-11:
Is it actually _possible_ to unmount the root directory `/` ? (even if you could, I guess it'd probably result in lots of other things breaking?)
Maybe a better fix would be to always explicitly exclude the drive that's mounted at `/` (e.g. make it a non-selectable "System Drive" or something, see https://github.com/balena-io-modules/drivelist/issues/126 ), instead of (mistakenly) labelling it as "Drive Contains Image" ?  IMHO "The image path starts with one of this drive mountpoints" isn't a very user-friendly message \*shrug\*

zvin, 2018-12-11:
@lurch 

>Is it actually possible to unmount the root directory /
This is not the question;

>Maybe a better fix would be to always explicitly exclude the drive that's mounted at /
This is already done, `/` will only be shown in unsafe mode.

>instead of (mistakenly) labelling it as "Drive Contains Image"
It is a system drive but it also contains the mountpoint of the drive that contains the image.

> IMHO "The image path starts with one of this drive mountpoints" isn't a very user-friendly message *shrug*
This is not the label that we'll use, we'll use "Drive Mountpoint Contains Image", see https://github.com/balena-io/etcher/pull/2582

closed 2019-04-03 by zvin