balena-io/etcher #2288: Cannot select drive in Etcher

witwtf, 2018-04-26:
- **Etcher version: 1.4.3** 
- **Operating system and architecture: Windows 10 64 bit** 
- **Image flashed: Retropie** 
- **Do you see any meaningful error information in the DevTools? No** 

I'm trying to flash Retropie onto an SD card however I cannot seem to select a drive.  Etcher sees my SD card however each time I click on the drive, a split second later it unselects itself.  I can keep clicking on it, however each time it will unselect itself in less than a second and so I cannot progress to flashing the SD.  I've tried to reformat the SD and tried using it elsewhere and the SD card works.  I've tried with the 32bit/64bit, portable and installer and even a different image.  It seems something in the programme keeps deselecting the drive?

Thanks.

H

lurch, 2018-04-26:
Duplicate of #2267 which is reportedly fixed in Etcher 1.4.4

witwtf, 2018-04-26:
Thanks for that!

closed 2018-04-26 by witwtf

jhermsmeier, 2018-04-27:
Duplicate of #2267