balena-io/etcher #2289: Etcher 1.4.4 Windows 64 executable trigger the Windows Defender smart screen on windows 10

chris282, 2018-04-27:
- **Etcher 1.4.4* 
- **Windows 10** 
- **NONE** 
- **Do you see any meaningful error information in the DevTools?** 

https://forums.resin.io/t/etcher-v1-4-4-release/3016

chris282, 2018-04-27:
[image]

lurch, 2018-04-27:
IIRC this always seems to happen whenever a new version of Etcher is released :-(
I think it goes away once more people start using Etcher 1.4.4 ?

jhermsmeier, 2018-04-27:
Correct; "SmartScreen Protection shows the above message when you try to run a **newly released program or an application that has not yet established a reputation.**"<sup>[[1]](https://blogs.msdn.microsoft.com/vsnetsetup/2013/01/09/windows-smartscreen-prevented-an-unrecognized-app-from-running-running-this-app-might-put-your-pc-at-risk/)</sup>

You can click on "More info" (or "Informations complémentaires" in your case) to see the publisher information (which should read "Resin, Inc.") and then run it.

L3gomancer, 2018-05-30:
chris282, Mac also has built in security software that [gives a warning](https://support.apple.com/kb/ph25088?locale=en_US) when installing software downloaded from the internet, so it's not a fault and it's not only Etcher.
Stop swearing at hard working developers of free software.

duzkiss, 2018-08-15:
Hello Jonas Hermsmeier...I know that Resin.io is now part of neverware is some way, this is a question about PowerPC.  Is there any way or any Chrome OS variant that will work on the PowerPC architecture?

Thank you

jhermsmeier, 2018-08-28:
@duzkiss I'm not sure what you mean by "Resin.io is now part of neverware", but Etcher is currently not available for Chrome OS or PowerPC due to technical limitations. As this question is unrelated to this issue, please open a new issue if you have further questions.

jhermsmeier, 2018-08-28:
Closing, as this is an operating system feature, not an Etcher bug

closed 2018-08-28 by jhermsmeier