balena-io/etcher #2303: gpg: keyserver receive failed: No data

NatoBoram, 2018-05-02:
- **Etcher version:** N/A
- **Operating system and architecture:** Ubuntu 18.04 LTS x86_64
- **Image flashed:** N/A
- **Do you see any meaningful error information in the DevTools?** N/A

```bash
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.0M3e6QLdQ9/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data
```

jhermsmeier, 2018-05-02:
Duplicate of #1786

closed 2018-05-09 by jhermsmeier

djkerya, 2019-01-10:
sudo apt-key adv --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.3wzLSL1SLd/gpg.1.sh --keyserver hkp://pgp.mit.edu:80 --recv-keys 379CE192D401AB61
gpg: keyserver receive failed: No data