balena-io/etcher #2304: Etcher is not working on Xubuntu 18.04

Valkia-INNOS, 2018-05-03:
Etcher v1.4.4 64bit
Xubuntu 18.04lts x86_x64
any iso image is not working (etcher doesn't run)

jhermsmeier, 2018-05-03:
Are you using the deb package, or the AppImage?

lurch, 2018-05-03:
Probably a duplicate of #1305 ?

nis, 2018-05-05:
It is a duplicate of #1305 
At least the solution is the same: `sudo apt-get install -y libgconf-2-4`
And then the AppImage will run. I'm on Ubuntu 18.04, had the same error, installed libgconf-2-4. Etcher is etching in the background as I type this.

lurch, 2018-11-06:
@diml923 If you read through the linked issue #1305 you can see all the gory low-level details :wink:

prashantjeena, 2019-06-10:
> It is a duplicate of #1305
> At least the solution is the same: `sudo apt-get install -y libgconf-2-4`
> And then the AppImage will run. I'm on Ubuntu 18.04, had the same error, installed libgconf-2-4. Etcher is etching in the background as I type this.

where do we have to run this command....coz on gitbash it's not running.

lurch, 2019-06-10:
@prashantjeena According to https://stackoverflow.com/questions/17807485/what-is-the-exact-meaning-of-git-bash gitbash is a Windows program. The instructions here relate to the Linux version of Etcher, not the Windows version.