balena-io/etcher #2312: Loading screen when Etcher Pro is updating

Shou, 2018-05-08:
Has to be a static page that fits the Etcher Pro screen (1024x600px).

Design is a pre-requisite.

lurch, 2018-05-15:
1024x600 ?! Wasn't Etcher only recently tweaked to be 800x480? Does that mean #2299 is out of date already? :wink:

konmouz, 2018-05-16:
cc @Shou 

[image]

jhermsmeier, 2018-05-16:
As this has to be a static *image*, rather than a page, I don't think we'll be able to display which version we're updating for now, as we'd have to render the image prior to updating for that.

Shou, 2018-05-16:
There's an updated version on Zeplin now. Looks good :)

curcuz, 2018-05-18:
@konmouz we need to get rid of the spinner - this is a static image

konmouz, 2018-05-23:
@curcuz, we removed the spinner in the new version.

[image]

konmouz, 2019-06-03:
Update screen design - @konmouz 
Implement screen - @thundron

konmouz, 2019-06-24:
[image]

https://zpl.io/VDKA3vw -- @thundron please update screen