balena-io/etcher #2321: Burn own backup (made with Win32DiskImager) image to 160Gb SSD

fritserasmus, 2018-05-09:
- **Etcher version:** 1.3.1
- **Operating system and architecture:** Win 7 64Bit
- **Image flashed:** Can not flash image
- **Do you see any meaningful error information in the DevTools?** 

**_Error Message:_
Attention
The writer process ended unexpectedly.
Please try again, and contact the Etcher team if problem persists.**

I have successfully flashed many other images to SD. This is the first one to SSD - **I did tick "Unsafe Mode"**
 Nope

jhermsmeier, 2018-05-09:
I see you're still using an older Etcher version, and we have since released Etcher v1.4.4 – can you try with that?

fritserasmus, 2018-05-09:
Could it be that I am trying to flash a self-created-img file using Win32DiskImager.
Reason I am asking is that I successfully flashed stretch and now re-working that.

lurch, 2018-05-14:
How big is the image file that you're attempting to flash? Is Win32DiskImager able to write your backup image successfully?

fritserasmus, 2018-05-22:
@jhermsmeier ,

Thanks very much for your reply.
It was as simple as you explained: update etcher to the latest and I successfully wrote the 32Gb image to the SSD and everything just runs very happily.

For anyone else doing the same: 
I then downloaded stretch image and burnt to another Micro SD and installed GPARTED .
With SSD installed and the MicroSD I started the RPi that then booted from the MicroSD and was able to make the partition as big as I liked!

closed 2018-05-22 by fritserasmus