balena-io/etcher #2334: Does Etcher expand the rootfs partition after writing the image?

beruic, 2018-05-11:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Ubuntu 16.04 x64
- **Image flashed:** Newest Raspbian lite
- **Do you see any meaningful error information in the DevTools?** Irrellevant

This is just a question.

I am asking because I cannot figure out why my root partition is expanded after writing the image. The Etcher documentation (AFAIK) mention nothing of this, the image is only 1.9 GB, and 4 GB is still the recommended minimum card size.

jhermsmeier, 2018-05-12:
Hey there, Etcher does not expand any filesystems / partitions after flashing – I think Raspbian automatically does that on first boot nowadays, but need to double-check that.

beruic, 2018-05-14:
That is strange, because it is expanded already even though I have just written the 1.9 GB image to the card.

jhermsmeier, 2018-05-14:
Did you flash it from the zip archive, or did you extract the image first? And to what size did it expand?

beruic, 2018-05-16:
I flashed from the zip to an 8 GB card.

lurch, 2018-05-16:
@beruic what tool are you using to determine that the image is being "pre-expanded" after being flashed with Etcher but before being booted in the RPi?

beruic, 2018-05-16:
I use the disk tool in Ubuntu.
Just tried with `dd` and the disk is not expanded. Trying Etcher with the uncompressed image.

beruic, 2018-05-16:
This is very odd. Now I have tried multiple times with both the zipped image and the not zipped image, and the partition is not expanded when I look at it after flashing. I must have been dreaming or something. Closing this issue.

closed 2018-05-16 by beruic

jhermsmeier, 2018-05-16:
That is odd indeed – well, should it happen again some time, feel free to reopen this!