balena-io/etcher #2335: E: Unable to locate package etcher-electron

ohprg, 2018-05-12:
- **Attempting Upgrade to v  1.4.3** 
- **Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                4
On-line CPU(s) list:   0-3
Thread(s) per core:    2
Core(s) per socket:    2
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 58
Model name:            Intel(R) Core(TM) i5-3320M CPU @ 2.60GHz
Stepping:              9
CPU MHz:               1497.888
CPU max MHz:           3300.0000
CPU min MHz:           1200.0000
BogoMIPS:              5188.03
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              3072K
NUMA node0 CPU(s):     0-3
========================================================
Here is what happened:
========================================================

*****@*****:~$ echo "deb https://dl.bintray.com/resin-io/debian stable etcher" | sudo tee /etc/apt/sources.list.d/etcher.list
[sudo] password for alan: 
deb https://dl.bintray.com/resin-io/debian stable etcher
*****@*****:~$ sudo apt-key adv --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
Executing: /tmp/apt-key-gpghome.Eky30ldLdG/gpg.1.sh --keyserver keyserver.ubuntu.com --recv-keys 379CE192D401AB61
gpg: key 379CE192D401AB61: "Bintray (by JFrog) <bintray@bintray.com>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
alan@SATTOP-02:~$ sudo apt-get update
Hit:1 http://security.debian.org/debian-security stretch/updates InRelease
Ign:2 http://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                                     
Hit:3 http://dl.google.com/linux/chrome/deb stable Release                                                                                                                                       
Ign:4 http://ftp.us.debian.org/debian stretch InRelease                                        
Hit:6 http://ftp.us.debian.org/debian stretch-updates InRelease          
Ign:7 http://ppa.launchpad.net/webcamstudio/webcamstudio-dailybuilds/ubuntu cosmic InRelease
Hit:8 http://ftp.us.debian.org/debian stretch Release                    
Ign:10 http://ppa.launchpad.net/webcamstudio/webcamstudio-stable/ubuntu cosmic InRelease
.............................................
.............................................
.............................................
.............................................
Ign:28 http://ppa.launchpad.net/webcamstudio/webcamstudio-stable/ubuntu cosmic/main DEP-11 128x128 Icons
Reading package lists... Done
E: The method driver /usr/lib/apt/methods/https could not be found.
N: Is the package apt-transport-https installed?
W: The repository 'http://ppa.launchpad.net/webcamstudio/webcamstudio-dailybuilds/ubuntu cosmic Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
W: The repository 'http://ppa.launchpad.net/webcamstudio/webcamstudio-stable/ubuntu cosmic Release' does not have a Release file.
N: Data from such a repository can't be authenticated and is therefore potentially dangerous to use.
N: See apt-secure(8) manpage for repository creation and user configuration details.
E: Failed to fetch https://dl.bintray.com/resin-io/debian/dists/stable/InRelease  
E: Failed to fetch http://ppa.launchpad.net/webcamstudio/webcamstudio-dailybuilds/ubuntu/dists/cosmic/main/binary-amd64/Packages  404  Not Found
E: Failed to fetch http://ppa.launchpad.net/webcamstudio/webcamstudio-stable/ubuntu/dists/cosmic/main/binary-amd64/Packages  403  Forbidden
E: Some index files failed to download. They have been ignored, or old ones used instead.
*****@*****:~$  sudo apt-get install etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package etcher-electron

Thoughts....
Alan

lurch, 2018-05-14:
Duplicate of #2144 ?

jhermsmeier, 2018-07-06:
You need to install `apt-transport-https`, in order to fetch from HTTPS repositories, as the error log notes:

```console
E: The method driver /usr/lib/apt/methods/https could not be found.
N: Is the package apt-transport-https installed?
```

closed 2018-07-06 by jhermsmeier

Bskysdz, 2021-06-25:
Sorry but this solution doesn't work for me :(

> You need to install `apt-transport-https`, in order to fetch from HTTPS repositories, as the error log notes:
> 
> ```
> E: The method driver /usr/lib/apt/methods/https could not be found.
> N: Is the package apt-transport-https installed?
> ```

View from my terminal:
```
root@ender5pro:~ # sudo apt install apt-transport-https
Reading package lists... Done
Building dependency tree       
Reading state information... Done
apt-transport-https is already the newest version (1.8.2.3).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
root@ender5pro:~ # curl -1sLf \
>    'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
>    | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=raspbian  version=10  codename=buster  arch=armv7l  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Importing 'balena/etcher' repository GPG key into apt ...
   OK: Checking if upstream install config is OK ...
   OK: Installing 'balena/etcher' repository via apt ...
   OK: Updating apt repository metadata cache ...
   OK: The repository has been installed successfully - You're ready to rock!

root@ender5pro:~ # sudo apt-get update
Hit:1 http://raspbian.raspberrypi.org/raspbian buster InRelease
Hit:2 http://archive.raspberrypi.org/debian buster InRelease
Get:3 https://dl.cloudsmith.io/public/balena/etcher/deb/raspbian buster InRelease [5,056 B]
Fetched 5,056 B in 2s (2,681 B/s)                 
Reading package lists... Done
root@ender5pro:~ # sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher-electron
root@ender5pro:~ # sudo apt-get install balena-etcher
Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package balena-etcher
```

lurch, 2021-06-28:
```
distro=raspbian  version=10  codename=buster  arch=armv7l  
```

See #2870