balena-io/etcher #2338: Support flashing images from tarballs

jhermsmeier, 2018-05-14:
Implement support for flashing images from .tar archives

lurch, 2018-05-14:
See also #1895

jdmarshall, 2019-11-03:
What ever became of https://github.com/balena-io/etcher/blob/bdf920f86459d4ab63e62bd3b380f6c2d336ceac/docs/EXTENDED-ARCHIVES.md ?

It looks like there was interest in providing metadata along with the images, and as I'm sorta keen to have some of the facilities that file hinted at.

It seems to me that if zip files are supported, pulling an img out of a tarball would represent feature parity with zip files.

If I tried to put together a PR to make compressed tarballs work like zip files, would you be amenable to landing it?

lurch, 2019-11-04:
IIRC Extended Archives became deprecated, but I _think_ parts of it are still supported, and e.g. https://www.ev3dev.org/downloads/ uses them.
And see https://github.com/balena-io/etcher/pull/707#issuecomment-274945531