balena-io/etcher #2357: Won't recognize SD card inserted into MacBook

volare71, 2018-05-18:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Macbook Pro OSX Yosimite 10.10.1
- **Image flashed:** Raspian_stretch.zip
- **Do you see any meaningful error information in the DevTools?**  Didn't look

 Macs do not have an Alt key??? fn, control,option

jhermsmeier, 2018-05-19:
Oh, the Alt key is also named the Option key on some Mac keyboards. We should add that to the instructions.

Did you see any errors pop up? And does it show up if you enable "Unsafe mode" through the settings?

volare71, 2018-05-19:
Thanks:
No error popped up and I did not enable unsafe mode.

vatbrain, 2018-05-25:
Having same issue with Etcher 1.4.4 on Macbook Pro with OSX 10.9.5
No error message other than "No removable drive detected".  
Wish I had not overwritten my previous Etcher version (1.4.2?), which was seeing the SD card.

lurch, 2018-05-25:
> Wish I had not overwritten my previous Etcher version

Old releases are available from https://github.com/resin-io/etcher/releases
I think if you can track down the last version in which it _did_ work, and the first version in which it _doesn't_ work, that would help the Etcher team narrow down the problem.

vatbrain, 2018-05-25:
@lurch Thank you for the link to Old releases.
1.4.3 and 1.4.1 did not recognize my SD card.
1.3.1 recognized the SD card.

jhermsmeier, 2018-06-06:
@volare71 thanks for checking!
@volare71, @vatbrain can you tell us which particular SD-card you've been using (name & vendor), and by which means they were connected?

@T-Firefly can you post the output of when you run `example/index.js` with the SD-card connected? That'd really help us figure this out

kriekprinsloo, 2018-06-06:
If my input is not unwelcome, I would like to say that the same thing
happens with my SD card reader.  I have found out that my late 2012 MacBook
Pro's SD drive is half broken.  You should try using a different computer,
see if your problem is fixed.

vatbrain, 2018-06-07:
@jhermsmeier I was using a SanDisk Extreme 16GB micro SD HC1 plugged into an adapter and then into the SD card slot on my 2010 macbook pro.

ThomasKaiser, 2018-07-06:
I'm having the same problem on my older MacBook that has an internal SD card reader. Affected version is 1.4.1, 1.4.3 and 1.4.4 while 1.3.1 works.

I'm using the CLI variant and ran `execsnoop -a` in parallel. Situation looks like this (`disk3` the SD card in question):

    bash-3.2# diskutil list
    /dev/disk0
       #:                       TYPE NAME                    SIZE       IDENTIFIER
       0:      GUID_partition_scheme                        *251.0 GB   disk0
       1:                        EFI EFI                     209.7 MB   disk0s1
       2:          Apple_CoreStorage                         250.1 GB   disk0s2
       3:                 Apple_Boot Recovery HD             650.0 MB   disk0s3
    /dev/disk1
       #:                       TYPE NAME                    SIZE       IDENTIFIER
       0:                  Apple_HFS MacOS X                *249.8 GB   disk1
                                     Logical Volume on disk0s2
                                     46978F99-92F7-4249-B00D-D8E5C6ACA1A9
                                     Unencrypted
    /dev/disk3
       #:                       TYPE NAME                    SIZE       IDENTIFIER
       0:     FDisk_partition_scheme                        *4.0 GB     disk3
       1:             Windows_FAT_32 NO NAME                 78.6 MB    disk3s1
       2:                      Linux                         3.8 GB     disk3s2

With 1.3.1 `diskutil info` is called constantly while with 1.4.1 and above it's `diskutil info -plist`:

1.3.1:

    root            74696   0,0  0,0  2469680   2672 s004  R+    6:08pm   0:00.01 diskutil list
    root            74707   3,7  0,0  2495128   3312 s004  U+    6:08pm   0:00.01 diskutil info /dev/disk0
    root            74707   3,7  0,0  2495128   3312 s004  U+    6:08pm   0:00.01 diskutil info /dev/disk0
    root            74766   0,0  0,0  2494604   3308 s004  S+    6:08pm   0:00.01 diskutil info /dev/disk1
    root            74766   0,0  0,0  2494604   3832 s004  S+    6:08pm   0:00.02 diskutil info /dev/disk1
    root            74766   7,6  0,0  2494604   3848 s004  S+    6:08pm   0:00.03 diskutil info /dev/disk1
    root            74832   0,2  0,0  2494604   3304 s004  S+    6:08pm   0:00.01 diskutil info /dev/disk3
    root            74832   0,2  0,0  2494604   3304 s004  S+    6:08pm   0:00.01 diskutil info /dev/disk3

1.4.x

    root            73944   0,0  0,0  2443268   2284 s004  R+    6:08pm   0:00.01 diskutil list -plist
    root            73944   0,0  0,0  2494604   3308 s004  S+    6:08pm   0:00.01 (diskutil)
    root            73951   2,6  0,0  2469680   2732 s004  S+    6:08pm   0:00.01 diskutil info -plist /dev/disk0
    root            73951   2,6  0,0  2494604   3316 s004  S+    6:08pm   0:00.02 diskutil info -plist /dev/disk0
    root            73958   0,0  0,0  2461488   2728 s004  S+    6:08pm   0:00.01 diskutil info -plist /dev/disk1
    root            73958   3,6  0,0  2494604   3320 s004  S+    6:08pm   0:00.01 diskutil info -plist /dev/disk1
    root            73958   3,6  0,0  2494604   3880 s004  S+    6:08pm   0:00.03 diskutil info -plist /dev/disk1
    root            73970   0,0  0,0  2423612     24 s004  S+    6:08pm   0:00.00 diskutil info -plist /dev/disk3

I would assume parsing the output goes wrong. Here's output from `diskutil info -plist /dev/disk3 2>&1 | curl -F 'f:1=<-' http://ix.io`: http://ix.io/1gfb

Please note that `diskutil info -plist` outputs to `stderr` while `diskutil info` outputs to `stdout`. So maybe a simple redirect `2>&1` will already fix it?

ThomasKaiser, 2018-07-06:
> Please note that diskutil info -plist outputs to stderr while diskutil info outputs to stdout.

Hmm... nope, tested again and that was wrong (obviously no output one time reported by `curl` which led to my wrong assumption). Anyway: attached an external card reader to my more recent MacBook and here 1.4.4 works.

Old MacBook:

    bash-3.2# sw_vers
    ProductName:	Mac OS X
    ProductVersion:	10.10.5
    BuildVersion:	14F2511

New MacBook (where 1.4.x works):

    bash-3.2# sw_vers
    ProductName:	Mac OS X
    ProductVersion:	10.13.5
    BuildVersion:	17F77

`diskutil info -plist /dev/disk2 2>&1 | curl -F 'f:1=<-' http://ix.io` (on 10.13 now): http://ix.io/1gfn

ThomasKaiser, 2018-07-06:
Other difference between my old MacBook and my new one (with external card reader attached):

Old (no `Removable*` occurences):

	<key>Internal</key>
	<true/>

New:

	<key>Internal</key>
	<false/>
	<key>Removable</key>
	<true/>
	<key>RemovableMedia</key>
	<true/>
	<key>RemovableMediaOrExternalDevice</key>
	<true/>

ThomasKaiser, 2018-07-08:
Took some time but now I did the obvious: Using my external card reader on my old MacBook (the one running with 10.10.5 with an internal reader). No problem with Etcher 1.4.x recognizing the card.

* External card reader: `diskutil info -plist /dev/disk2 2>&1 | curl -F 'f:1=<-' http://ix.io`: http://ix.io/1gqI
* Internal card reader: `diskutil info -plist /dev/disk3 2>&1 | curl -F 'f:1=<-' http://ix.io`: http://ix.io/1gqJ

So now it seems to me that the problem with 1.4.x only affects the internal card reader of older MacBooks? BTW: I really like the internal card reader since [my external reader is crap](https://forum.armbian.com/topic/7360-burn-iso-image-to-micro-sd-card/?do=findComment&comment=57564).

ThomasKaiser, 2018-07-08:
> Etcher actually uses https://github.com/resin-io-modules/drivelist to list the drives

Thanks for the hint.

By looking at http://ix.io/1gqJ and http://ix.io/1gfb I would assume [this](https://github.com/resin-io-modules/drivelist/blob/39b988eb58dbd7fb4b7930a0c02c106b943d6012/lib/diskutil.js#L97) is the culprit (interpreting SD cards in a MacBook's internal slot being system disks?)

jhermsmeier, 2018-07-08:
Thanks @ThomasKaiser, that data will be very helpful!

shawaj, 2020-12-13:
@ThomasKaiser @volare71 can you test on the latest version v1.5.112?

ThomasKaiser, 2020-12-14:
[Nope](https://www.cnx-software.com/2020/09/08/usbimager-a-lightweight-alternative-to-balenaetcher/#comment-575334)

shawaj, 2020-12-14:
@ThomasKaiser does your "Nope" mean you won't try it or you have tried it and it isn't working?

ThomasKaiser, 2020-12-14:
@shawaj I neither use the laptop from 2.5 years ago nor Etcher any more. So no testing from my side.

shawaj, 2020-12-16:
@ThomasKaiser thanks for the clarification 👍

closed 2023-07-12 by dfunckt