balena-io/etcher #2359: Etcher 1.4.4 does not run on Ubuntu budgie 18.04 minimal 64bit

Der-Bauer, 2018-05-20:
- **Etcher version:1.4.4** 
- **Operating system and architecture:Ubuntu Budgie 18.04 minimal 64Bit** 
- **Image flashed:none** 
- **Do you see any meaningful error information in the DevTools?** 
      First i got problems to intergrated etcher to my menu after resolved this.I want to  start ether and 
     seems it´s  running but some seconds later it closed its self

lurch, 2018-05-20:
Duplicate of #1305 ?

Der-Bauer, 2018-05-20:
It is
-- 
Diese Nachricht wurde von meinem Android Mobiltelefon mit GMX Mail gesendet.

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "37567f93-e2a7-4e2a-ad37-a9160fc62647",
"title": "Re: [resin-io/etcher] Etcher 1.4.4 does not run on Ubuntu budgie 18.04 minimal 64bit  (#2359)",
"sections": [
{
"text": "",
"activityTitle": "**Andrew Scheller**",
"activityImage": "https://cloud.githubusercontent.com/assets/143418/15842166/7c72db34-2c0b-11e6-9aed-b52498112777.png",
"activitySubtitle": "@lurch",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"resin-io/etcher\",\n\"issueId\": 2359,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"resin-io/etcher\",\n\"issueId\": 2359\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/resin-io/etcher/issues/2359#issuecomment-390493275"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 337140153\n}"
}
],
"themeColor": "26292E"
}

lurch, 2018-06-05:
@jhermsmeier This can be closed as a duplicate.

jhermsmeier, 2018-06-05:
Duplicate of #1305

closed 2018-06-05 by jhermsmeier