balena-io/etcher #2361: Error when running first time on ubuntu

mskogly, 2018-05-21:
- **Etcher version:** 
1.0.0-beta.19
- **Operating system and architecture:** 
Ubuntu 16.04 lts
- **Image flashed:** 
None, error comes up when running Etcher. 
- **Do you see any meaningful error information in the DevTools?** 

TypeError: Invalid Version: static
    at new SemVer (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:293:11)
    at compare (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:566:10)
    at Function.gt (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/node_modules/semver/semver.js:595:10)
    at bucketEntries.reduce.latest (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:93:20)
    at Array.reduce (native)
    at /tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:87:43
    at Parser.<anonymous> (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:484:18)
    at emitOne (events.js:96:13)
    at Parser.emit (events.js:188:7)
    at Object.onclosetag (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:445:26)
    at emit (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:640:35)
    at emitNode (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:645:5)
    at closeTag (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:905:7)
    at Object.write (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/sax/lib/sax.js:1449:13)
    at Parser.exports.Parser.Parser.parseString (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:503:31)
    at Parser.parseString (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:7:59)
    at Object.exports.parseString (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/xml2js/lib/xml2js.js:535:19)
    at /tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/etcher-latest-version/index.js:80:9
    at $http.get.then (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/lib/gui/components/update-notifier/services/update-notifier.js:53:18)
    at processQueue (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/angular/angular.js:16648:37)
    at /tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/angular/angular.js:16692:27
    at Scope.$eval (/tmp/.mount_BBkmCV/usr/bin/resources/app.asar/node_modules/angular/angular.js:17972:28)

lurch, 2018-05-21:
> Etcher version: 1.0.0-beta.19

Umm???? You _do_ realise that that's a super-old version?! https://etcher.io currently has v1.4.4 for download.

Shou, 2018-05-22:
Yeah, I would recommend upgrading :)

closed 2018-10-15 by jviotti