balena-io/etcher #2365: Etcher installer takes too much space (over 70 MB for macOS)

magicgoose, 2018-05-24:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** macOS, x86_64
- **Image flashed:** None yet
- **Do you see any meaningful error information in the DevTools?** None yet

I just wanted to say that the installer size seems almost absurdly big for an ISO writing utility. And it takes non-negligible time to download. Considering that for many scenarios this is a tool that's going to be used only once, that's a lot of network traffic and this should be improved.

For comparison, unetbootin weighs less than 9 MB. That's more than 7 times less than this.

lurch, 2018-05-24:
I've seen people raise this 'concern' a few times before. Etcher contains much more functionality than just a 'simple ISO writer' see e.g. https://resin.io/blog/etcher-now-with-multi-write-and-compute-module-support/

IIRC a lot of the overhead in Etcher's download-size is due to it using the cross-platform Electron framework https://github.com/resin-io/etcher/blob/master/docs/ARCHITECTURE.md

> that's a lot of network traffic

But it's presumably still fairly small when compared to the actual ISO that you're going to be downloading? :wink:

magicgoose, 2018-05-24:
@lurch not all ISOs are big, some of them might be rather small, for example GParted (318 MB) which is actually useful in some real world situations and 77.4/(77.4 + 318) ≈ 0.2, that is, Etcher would be roughly 20% of the total network traffic in this case, I think it's not "fairly small". There are also other distributions which have little embedded packages…  
You might kind of have a point for really big ISO files like the Ubuntu which is almost over 2 GiB (if it's not a netinstall image), though.

> IIRC a lot of the overhead in Etcher's download-size is due to it using the cross-platform Electron framework https://github.com/resin-io/etcher/blob/master/docs/ARCHITECTURE.md

I am quite confident that this is the main reason. This still needs to be solved somehow, IMO. As now this tool is being suggested as the recommended way to write ISO images to USB sticks at 
`tutorials.ubuntu.com`, so this will inevitably affect a lot of people.

jhermsmeier, 2018-06-05:
As @lurch mentioned, about 80% of the size is due to Electron, and a further expansion in size is due to the way it's packaged for macOS. While we're already looking at reducing the size of Etcher itself, it'l still not have a very significant impact.

Closing this as wontfix for the time being due to the reasons outlined above; feel free to re-open if you feel otherwise :)

closed 2018-06-05 by jhermsmeier

magicgoose, 2018-06-05:
@jhermsmeier 
> While we're already looking at reducing the size of Etcher itself, it'l still not have a very significant impact.

The impact will be more significant if Electron is replaced with something else.

jhermsmeier, 2018-06-05:
Sure, but there's a distinct lack of "something else"'s with the features and capabilities we're looking for. Most notably the cross-platform aspect. Rewriting it in "native framework X" isn't really an option  either (yes, we have thought about that as well), as that'd easily triple the work required to make this work everywhere, and is also plagued by incompatibilities.

magicgoose, 2018-06-05:
> Most notably the cross-platform aspect.

Qt is cross-platform.

lurch, 2018-06-05:
@magicgoose Etcher has been in development for 2 and a half years now - I don't think they'll be rewriting the entire UI in a different framework just because some people want a smaller download. (apologies if that sounds a bit blunt)

But I believe the Etcher team are planning to release an "Etcher SDK" of some sort, which AIUI would make it easier for you to write your own UI (in the GUI framework of your choice?) on top of the underlying SDK.

My1, 2019-12-18:
if not already done, how about only including the parts of electron you actually use?

ruslang02, 2021-02-02:
Rewriting the project to [NodeGui](https://github.com/nodegui/nodegui) would **significantly** decrease the end-product's size, not comprimising on compatibility. @nodegui