balena-io/etcher #2366: Etcher erroneously thinks micro SD card is locked. Cannot write to it.

davidrnewman, 2018-05-24:
When I try to flash a micro SD card for my Pi-Top, Etcher won't let me select the card. It claims the card is locked. There is no lock on a micro SD card. I have just plugged it into an adaptor, and that into a full size SD card slot on my computer.

- **Etcher version: etcher-electron-1.4.4-x86_64.AppImage
- **Operating system and architecture: Kubuntu Linux 18.04 x86_64 (on AMD Ryzen)
- **Image flashed: It didn't.
- **Do you see any meaningful error information in the DevTools? No

kriekprinsloo, 2018-05-24:
Yes, this has happened to me before as well.  It turned out that there was
something wrong with my SD card reader.  Maybe try a different computer?

Also, there is a lock usually on your microSD card adapter.  Try a few
other adapters, see if that fixes the problem.

jhermsmeier, 2018-05-25:
Thanks for helping out here @kriekprinsloo! Going to close this then, as the read-only switch was at fault here ;)

closed 2018-05-25 by jhermsmeier

kriekprinsloo, 2018-05-25:
No problem :) glad to be able to help you guys out.

nalply, 2019-01-24:
Another tip: for me it helped to tap the adapter with the finger really hard, sometimes the lock state seems to get stuck