balena-io/etcher #2378: Recover the data in the USB drive

seevider, 2018-05-31:
Hi all,

I used the etcher to flash an os image to my USB drive but I forgot to copy all the original files in the USB drive before I did the action. Then, I lost all my files......

Is it possible to recover the USB drive? How can I do that?

Thanks

KeithIMyers, 2018-05-31:
Not likely

jhermsmeier, 2018-06-05:
Sorry to hear that @seevider, unfortunately that's bad news. There might be a possibility to recover some files, depending on large the image you flashed was, and some other factors. But I'm afraid we can't do much, apart from suggesting recovery software that helped some people before (have to go look them up again)

seevider, 2018-06-06:
Thank you for your information @jhermsmeier. I formatted my USB driver and the storage space went back to original size. I will try the recovery software to recover the data.

lurch, 2018-06-06:
Unfortunately formatting your drive first may make the recovery process less successful :-(

closed 2018-10-15 by jviotti