balena-io/etcher #2379: Choosing target UI isn't fail-safe enough

paul-hammant, 2018-06-02:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Mac, 64bit, HighSierra, latest patches applied.
- **Image flashed:** Raspbian from April.
- **Do you see any meaningful error information in the DevTools?** n/a

### Software would be even better if:

I had a sparse bundle already open when I opened Etcher. My intention was to blast Raspbian to a large SD card. The sparse bundle is pre-selected - eww!  Even when I open that mini-dialog and select the SD card, the sparse budndle by default is not de-selected.  Wow. UI affordances like this should be single select (with no default when there's a choice of two). Multiple selection should not be the default functionality of such a control. Sure, run with the green checkbox (rather than shift-select for multiple), but single select with no pre-choice if two is correct.

lurch, 2018-06-02:
Sounds like selecting a "sparse bundle" may be a bug in `drivelist` ?
Did you insert the SD card before or after starting Etcher? If you inserted the SD card _after_ starting Etcher, then Etcher probably only detected one available "writable drive" when it started up, and so auto-selected that single drive by default.

@jhermsmeier A thought that's just occurred to me - for advanced users, might it be worth adding a `noAutoSelection` to the settings? (which would obviously be false by default / if not set)

paul-hammant, 2018-06-03:
Autoselection is only half of my observation. The other half is that it should (by default) be a single selection list box. Selecting B unselects A, Selecting A after that, unselects B.

jhermsmeier, 2018-06-05:
@lurch yeah, we'll be exposing most settings in the UI soon-ish

jhermsmeier, 2018-06-05:
@paul-hammant I agree, the current behavior isn't ideal – iirc we even had multi-selection with <kbd>Shift</kbd> only at one point, but it might not have been obvious enough that it'd work from the UI at this time, and also would fail on touch screens.

@konmouz what do you think about reverting to having to hold <kbd>Shift</kbd> for multi-select to work, and displaying a little help message in the device selection saying something like "Use Shift to select multiple targets"? Of course we'd have to turn that off for touch screens...

lurch, 2018-06-05:
Just to be pedantic, if you wanted to mimic the behaviour of e.g. Windows Explorer, would it be better to use `Ctrl` for multi-selection? (as `Shift` is normally used for range-selection) :man_shrugging:

jhermsmeier, 2018-06-05:
> to use Ctrl for multi-selection? (as Shift is normally used for range-selection)

Ah yes, <kbd>Ctrl</kbd> it is, but I guess also having <kbd>Shift</kbd> for range selection wouldn't hurt in situations where you'd want to select a large number of devices

> I was wondering if it'd be worth adding a noAutoSelection setting so that users could totally disable auto-selection if they wanted to

Yeah, I got that – it'll end up in the swamp of settings we'll be exposing (as the opposite of that exists already)

konmouz, 2018-06-11:
@jhermsmeier, this won't be clear for the user. The issue here is that we shouldn't autoselect a drive when more than one options are available. If only one drive is detected and therefore autoselected, and then afterwards a second drive is available, we can deselect the already autoselected drive. In addition, the CTA indicates the amount of selected drives to inform the user about the total number of drives that are going to be flashed.

lurch, 2018-06-11:
> If only one drive is detected and therefore autoselected, and then afterwards a second drive is available, we can deselect the already autoselected drive.

That might be tricky, since AFAIK Etcher doesn't keep track of whether a drive was auto-selected or manually selected. If a user manually selects a drive, and then inserts another drive, they might not expect the first drive to be automatically unselected? (IYSWIM)

JeremyBoden, 2018-08-05:
I would.

jviotti, 2018-10-15:
Improved on v2 design

closed 2018-10-15 by jviotti