balena-io/etcher #2382: Getting error when trying to do flash write of raspberry binary zip.

vkg70, 2018-06-08:
- **Etcher version:** Version 1.4.4 (1.4.4)
- **Operating system and architecture:** Mac OS 10.13.5
- **Image flashed:** 

The elevated process died unexpectedly
The process error code was 126

- **Do you see any meaningful error information in the DevTools?** 

Thu Jun 07 2018 21:25:49 GMT-0700 (PDT) Restart after failure
gui.js:105081 etcher:sdk:scanner stop +1m true
gui.js:105081 Thu Jun 07 2018 21:27:08 GMT-0700 (PDT) Flashing 0 devices, 0% at 0 MB/s (total 0 MB/s) eta in undefineds with 0 failed devices
gui.js:105081 Thu Jun 07 2018 21:27:08 GMT-0700 (PDT) Flash ({"image":"/Users/tanishagupta/Downloads/2018-04-18-raspbian-stretch.zip","drives":["/dev/disk4"],"uuid":"2b63c99c-84c6-4e31-9d62-e67d2b86839a","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
gui.js:105081 Terminating IPC server
gui.js:105081 Thu Jun 07 2018 21:27:08 GMT-0700 (PDT) Flash error ({"error":{"name":"Error","message":"The elevated process died unexpectedly","description":"The process error code was 126","stack":"Error: The elevated process died unexpectedly\n    at Object.exports.createError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:292:18)\n    at sudoPrompt.execAsync.then.catch.catch (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/permissions.js:211:18)\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at catch_filter.js:24:41\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:681:18)\n    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>","report":false},"image":"/Users/tanishagupta/Downloads/2018-04-18-raspbian-stretch.zip","drives":["/dev/disk4"],"uuid":"2b63c99c-84c6-4e31-9d62-e67d2b86839a","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Thu Jun 07 2018 21:27:08 GMT-0700 (PDT) Open modal ({"name":"warning"})
gui.js:105081 {"stack":"Error: The elevated process died unexpectedly\n    at Object.exports.createError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:292:18)\n    at sudoPrompt.execAsync.then.catch.catch (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/permissions.js:211:18)\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at /Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/catch_filter.js:24:41\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:681:18)\n    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:17:14)\n    at <anonymous>","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false}
console.(anonymous function) @ gui.js:105081
gui.js:105081 etcher:sdk:scanner start +11s true
gui.js:105081 Thu Jun 07 2018 21:27:21 GMT-0700 (PDT) Modal accepted ({"name":"warning","value":true})
gui.js:105081 Thu Jun 07 2018 21:27:21 GMT-0700 (PDT) Restart after failure
gui.js:105081 Thu Jun 07 2018 21:27:48 GMT-0700 (PDT) Open modal ({"name":"drive-selector"})
gui.js:105081 Thu Jun 07 2018 21:27:50 GMT-0700 (PDT) Toggle drive ({"drive":{"size":63864569856,"isVirtual":false,"enumerator":"diskutil","logicalBlockSize":512,"raw":"/dev/rdisk4","error":null,"isReadOnly":false,"displayName":"/dev/disk4","blockSize":512,"adapter":"blockdevice","isSCSI":false,"isRemovable":true,"device":"/dev/disk4","busVersion":null,"isSystem":false,"busType":"USB","isCard":null,"isUSB":true,"mountpoints":[{"path":"/Volumes/RASPBIAN","label":"RASPBIAN"}],"description":"APPLE SD Card Reader Media","isUAS":null},"previouslySelected":false})
gui.js:105081 Thu Jun 07 2018 21:27:53 GMT-0700 (PDT) Modal accepted ({"name":"drive-selector"})
gui.js:105081 etcher:sdk:scanner stop +35s true
gui.js:105081 Thu Jun 07 2018 21:27:54 GMT-0700 (PDT) Flashing 0 devices, 0% at 0 MB/s (total 0 MB/s) eta in undefineds with 0 failed devices
gui.js:105081 Thu Jun 07 2018 21:27:54 GMT-0700 (PDT) Flash ({"image":"/Users/tanishagupta/Downloads/2018-04-18-raspbian-stretch.zip","drives":["/dev/disk4"],"uuid":"44ccf335-1d27-43b7-aceb-969dcbe07ec7","unmountOnSuccess":true,"validateWriteOnSuccess":true})
gui.js:105081 Elevating command: /Applications/Etcher.app/Contents/Frameworks/Etcher Helper.app/Contents/MacOS/Etcher Helper /Applications/Etcher.app/Contents/Resources/app.asar
gui.js:105081 Terminating IPC server
gui.js:105081 Thu Jun 07 2018 21:27:54 GMT-0700 (PDT) Flash error ({"error":{"name":"Error","message":"The elevated process died unexpectedly","description":"The process error code was 126","stack":"Error: The elevated process died unexpectedly\n    at Object.exports.createError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:253:17)\n    at Object.exports.createUserError (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/errors.js:292:18)\n    at sudoPrompt.execAsync.then.catch.catch (/Applications/Etcher.app/Contents/Resources/app.asar/lib/shared/permissions.js:211:18)\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at catch_filter.js:24:41\n    at tryCatcher (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/util.js:16:23)\n    at Promise._settlePromiseFromHandler (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:504:31)\n    at Promise._settlePromise (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:561:18)\n    at Promise._settlePromise0 (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:606:10)\n    at Promise._settlePromises (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/promise.js:681:18)\n    at Async._drainQueue (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:138:16)\n    at Async._drainQueues (/Applications/Etcher.app/Contents/Resources/app.asar/node_modules/bluebird/js/release/async.js:148:10)\n    at Async.drainQueues (/Applications/Etcher.app/Co [... cut ...] s/release/async.js:17:14)\n    at <anonymous>\n    at process._tickCallback (internal/process/next_tick.js:109:7)","message":"The elevated process died unexpectedly","description":"The process error code was 126","report":false}
console.(anonymous function) @ gui.js:105081
gui.js:105081 etcher:sdk:scanner start +4m true
gui.js:105081 Thu Jun 07 2018 21:32:09 GMT-0700 (PDT) Modal accepted ({"name":"warning","value":true})
gui.js:105081 Thu Jun 07 2018 21:32:09 GMT-0700 (PDT) Restart after failure

jhermsmeier, 2018-06-08:
Thanks @vkg70! This is odd. Does this happen with any image / target device, or only with this combination?

vkg70, 2018-06-08:
Haven't tried any other image. Does the target device have to be formatted with a specific filesystem? FAT32 etc?

jhermsmeier, 2018-06-08:
Ah okay. No, it shouldn't matter in which state the target device is. I haven't been able to reproduce this so far. Did you install Etcher from the dmg or zip archive?

closed 2018-10-15 by jviotti