balena-io/etcher #2387: 2 USB Drives bricked

jonaspauleta, 2018-06-11:
- 1.4.4
- Windows 10 64-bit
- ubuntu-18.04-desktop-amd64
- No

Hey guys,
I am very angry because I had 2 USB pen drives, from kingston, one was 16GB the other was 32GB, I have already used this USB drives for a long, and I decided to flash ubuntu on one drive and install it on my second SSD, but, guess what, it corrupted my usb drive. and then I flashed again in the other because i didn't know it was etcher causing that awful problem and cabum 2 usb drives corrupted, so, who have the responsability? I not for sure, because I haven't seen nothing warning me that my drive could brick when i click flash.

I have tryed a lot to recover but nothing, as a programmer, I would never release a software that blows hardware, and I have seen here in reedit that I am not the first one having this problem.

I hope you guys solve me this, because I want my 2 pen drives back.

And as I said, for the much you can try, there is no way back, it bricked.

[image]

lurch, 2018-06-11:
https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives :grey_question: 

Also, just because Windows can't read the contents of the USB drive, it doesn't mean that Ubuntu _wasn't_ written to the drive, because (unfortunately) Windows can't read Linux filesystems.

cj2tech, 2018-06-12:
I’ve had this issue I used as card formatted and uncheck quick format(to write every byte to 0). And works fine again.

jhermsmeier, 2018-06-12:
@jonaspauleta any luck with what @lurch pointed to, or @cj2tech suggested?

Apachez-, 2018-06-18:
A way to reset an USB-drive is to use something like this:

1) Run linux, if you dont then download live boot cd of Ubuntu and boot on that.

2) After the above booted plugin your USB-drive and find out which device id it receives through dmesg along with ls -la /dev/sd*

A hint here is to do the above before plugin the drive and then again after you did this to better see which id it picked up.

3) Now wipe the content of the drive (make sure you select the correct device id, if unsure then unplug your harddrive before booting that live cd so you dont wipe the wrong device) by this command:

>sudo dd if=/dev/zero of=/dev/sdx bs=1M count=10

Note that /dev/sdx is the device id we located in step 2 above.

4) Remove the drive from the computer and reattach it 2 or more seconds later.

5) Now you should be able to flash it again. If you want to use it as a regular FAT32 device then continue with below:

5.1) Locate the device id the drive got and partition it with FAT32 using cfdisk:

>sudo cfdisk /dev/sdx

In cfdisk remove any partitions seen, create a new one (if asked then select dos), make it all available size (should be default), select primary. Now you see a linux partion, select type and scroll until you see 0c W95 FAT32 (LBA) (sometimes its just c and not 0c).

Select Write to save the changes.

5.2) When getting back to your terminal unplug and reattach the drive.

5.3) Now create the filesystem by running:

>sudo mkfs.vfat /dev/sdx

Now you should have a working FAT32 drive.

Im not sure if you need to do all the unplug/reattach but it seems that some linux versions cache the (old) filesystem and this is only reset when you physically unplug/reattach the drive.

ralorimer, 2018-08-04:
Just for the record (i.e. so the developers are aware of another failure), a brand new PNY 32GB Flash Drive (literally removed from the packaging for this purpose) was also bricked by Etcher. 

I used Etcher to burn a ZorinOS iso to the drive (as described on the Zorin installation page). The burning was done on a ASUS Zen AIO desktop running Windows 10 (fully up to date v1803). The installation of Zorin was done on an old Acer Aspire (dual boot with Win10 v1803).

After doing this work I plugged the drive into the ASUS and received two popups saying drives E: and F: were not recognised and had to be formatted. (Both these drives related to the one PNY flash drive, not sure why, my technical knowledge on drive partitioning is somewhat limited. It must be something Etcher did as the drive was new and empty).

I was unable to continue in any way with the drive until I had formatted both E and F. After doing this, the drive was simply ignored by all operating systems on both computers. Plugging it in I got the Windows ding-dong sound but no drive was listed at all.

I tried various solutions found around these forums before I found the last post on this page advising to run "diskmgmt.msc" and fortunately this resolved the issue with no further problems.

Just thought I should post here because if I was a dev with Etcher I would be extremely concerned that the software is bricking so many drives.

lurch, 2018-08-06:
As mentioned before, AFAICT it's not "Etcher bricking drives", it's Windows being unable to read the filesystems used in the (Linux) images people are writing to their USB drives... :confused: 

> ... fortunately this resolved the issue with no further problems.

Ah, so your drive wasn't _actually_ "bricked" then? Something being bricked usually means it's totally broken and unrepairable.

ralorimer, 2018-08-06:
So, that being the case, why did Linux also not recognise the drive? Additionally, I have two other USB flash drives with Linux boot images on (made by someone else with a different system, don't know what) that are readable by Windows (or at least the boot files are). 

Bricking means to make something unusable (ie might as well be a brick) but this can be a temporary situation if you're fortunate enough to find a resolution.

lurch, 2018-08-06:
> So, that being the case, why did Linux also not recognise the drive? Additionally, I have two other USB flash drives with Linux boot images on...

See https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#flashing-ubuntu-isos

Different Linux distros probably set up their boot media in entirely different ways.

> Bricking means to make something unusable (ie might as well be a brick) but this can be a temporary situation

Ahh, I'd always considered "bricking" to be a more permanent thing. :man_shrugging:

lurch, 2018-08-06:
Of course, I might be **entirely** wrong and there might be some small error you're triggering in Etcher which the developers are as yet unaware of. If that's the case, I'm sure @jhermsmeier would love to hear about a reproducible test-case...

AWulkan, 2018-08-10:
Tried putting Ubuntu 18.04.1 on it and the USB became unusable.
Then I did what it said [here](https://github.com/resin-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives), but it didn't work...

```cmd
DISKPART> list disk

  Disk ###  Status         Size     Free     Dyn  Gpt
  --------  -------------  -------  -------  ---  ---
  Disk 0    Online          238 GB      0 B
  Disk 1    Online          465 GB   465 GB
  Disk 2    No Media           0 B      0 B

DISKPART> select disk 2

Disk 2 is now the selected disk.

DISKPART> clean

Virtual Disk Service error:
There is no media in the device.
```

The disk is a 16GB Sandisk USB 3.0.

closed 2018-10-15 by jviotti

techsolveprac, 2019-03-01:
Your flash drive can be recovered, read here --> https://www.techsolveprac.com/repair-flash-drive/

mirzaatifbaig, 2019-03-17:
I have the same issue as above 0B But the drive doesn't show anywhere and diskpart gives the error again and again and I Blame etcher for ruining my 5$ device

yashguptaz, 2019-10-21:
I found a weird solution to this problem.
The above solution didn't work for me.

1. Download any Linux distro which you can flash with Rufus like ElementaryOS.
2. Flash it to your USB Drive using Rufus.
3. Format your USB Drive as normal.

It miraculously worked for me.

lofcz, 2019-10-23:
Using Rufus to flash another / same distro, then formating resolved the problem for me