balena-io/etcher #2389: I am having trouble downloading the latest etcher version for windows 64

ellafairhurst, 2018-06-12:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **Do you see any meaningful error information in the DevTools?**

jhermsmeier, 2018-06-12:
Hi @ellafairhurst, can you describe the issue you're having in a little more detail? Does the download not complete, is it slow, or what's happening?

jhermsmeier, 2018-06-12:
And from where are you downloading it (https://etcher.io, github releases, etc...)?

ellafairhurst, 2018-06-12:
Actually, it seems to be working now, thanks!

jhermsmeier, 2018-06-12:
Alright, good to hear! Do let us know if you have any further troubles!

closed 2018-06-12 by jhermsmeier

ellafairhurst, 2018-06-23:
Hi,
    Have attached the zero cam to the pi zero, it lights up green, but
there is no red light on the zero cam, do you know why this would be?
Thanks,
            Ella

lurch, 2018-06-25:
That sounds like a question for https://www.raspberrypi.org/forums/ rather than the Etcher issues list :wink:

ellafairhurst, 2018-06-25:
I've downloaded etcher but can't download naturecam file, seems to be
corrupted.
Any suggestions?
Thanks
       Ella