balena-io/etcher #2390: Unable to select a 64GB SD Card

Dudematters, 2018-06-13:
- **Etcher version: 1.4.3** 
- **Operating system and architecture: MacOS 10.13.5** 
- **Image flashed: volumio-2.389-2018-03-26-pi.img.zip** 

I am trying to flash an 64GB SD Card for a raspberry pi with volumio. Ether sees the SD card but won't let me select it. It selects it for a brief moment but deselects it directly. I can format it on my Mac but nothing helps to make Etcher accept my card.

[image]

jhermsmeier, 2018-06-13:
Hey there, that's a bug we fixed in version 1.4.4; I'd recommend to update & try again

jhermsmeier, 2018-06-13:
Duplicate of https://github.com/resin-io/etcher/issues/2267

jhermsmeier, 2018-06-13:
Closing as duplicate

closed 2018-06-13 by jhermsmeier