balena-io/etcher #2391: /!\ Urgent /!\ 32 GO lost ...

Hadesa, 2018-06-13:
I lost my 32 GB that I had on my micro SD card because of your Flash software. How to recover them?

cj2tech, 2018-06-13:
I would suggest downloading SD card formatter (https://www.sdcard.org/downloads/formatter_4/index.html) or Rufus (https://rufus.akeo.ie.).

I would also suggest once done format using your Os built in formatted and unchecking quick format or set all bytes to 0. It will take a long time to do it but it will help clear out your SD of any bugs and set it to a factory state.

cj2tech, 2018-06-13:
I’d also reccoment to uninstall etcher.io and reinstall. And check to make sure all you drivers are up to date.

Hadesa, 2018-06-13:
I try formatter via the console

jhermsmeier, 2018-06-13:
@Hadesa we can't really help you without a little bit more info; what steps did you take until this occured? i.e. did you flash the card, and with which image, which version of Etcher are you running, etc.?

Hadesa, 2018-06-14:
Forget I managed it alone thanks to other forum

closed 2018-06-14 by jhermsmeier