balena-io/etcher #2392: Cannot Install Etcher

SwiftHackerz, 2018-06-15:
- **Etcher version: 1.4.4** 
- **Operating system and architecture: Windows x64** 
- **Image flashed: None** 

Whenever I install Etcher, my antivirus always scans the exe file and it says that there is a virus inside the program installer. And whenever I try to restore the installer, it always gives me a warning that there is a heuristic virus (Heur.AdvMLC) and it could harm my system. Is this a normal thing or is my computer reacting to this virus embedded in the installer? 

[image]

Synaxis, 2018-06-17:
https://www.virustotal.com/en/file/432203b640ace82b91a951146bd4d45fdc60ed0d9e91140abfa275a2108bee6c/analysis/

jhermsmeier, 2018-06-18:
@SwiftHackerz can I ask where you're downloading Etcher from?

SwiftHackerz, 2018-06-19:
I installed Etcher straight from the source. (https://etcher.io/)

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [resin-io/etcher] Cannot Install Etcher (#2392)",
"sections": [
{
"text": "",
"activityTitle": "**Jonas Hermsmeier**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@jhermsmeier",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"resin-io/etcher\",\n\"issueId\": 2392,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"resin-io/etcher\",\n\"issueId\": 2392\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/resin-io/etcher/issues/2392#issuecomment-398063523"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 346525028\n}"
}
],
"themeColor": "26292E"
}

closed 2018-10-15 by jviotti