balena-io/etcher #2394: Etcher causes the Windows explorer to not refresh automatically

FetzenRndy, 2018-06-17:
- **Etcher version:** 1.4.4
- **Operating system and architecture:** Windows 10 x64 
- **Image flashed:** DietPi Raspberry 3
- **Do you see any meaningful error information in the DevTools?** No

Every time I use Etcher my desktop stops refresehing automatically. When I create a folder ( That folder is not in any way related to Etcher ) the folder does not appear on the desktop until I manually refresh the dekstop / explorer / folder. Also Etcher wont flash the sd-card and is stuck in "Starting..."

I know for certain this is a Etcher problem. It only start to appear after first opening etcher and only goes away after **several** restarts.

I have no clue how Etcher would even cause such an issue. I had the same issue about 5 months ago and 
tried everything to get rid of the "not-refreshing-bug", the only thing that seemed to fix it was restarting the computer several times.

The other wired thing is that etcher works fine with no issues on my parents PC.

I open this Issue now, since I now reset my pc ( not because of Etcher ) and tried etcher again, and I still have this exact issue.

kenzierocks, 2019-01-20:
I have a similar issue -- Etcher seems to totally destroy Windows 10's ability to do anything on my end. Disk had been speedy, but the second I start flashing with Etcher Windows starts slowing down disk operations to nearly inoperable levels.

thundron, 2019-01-21:
We're investigating a bit on this, we have another issue about some users reporting unusually high CPU loads so this might be connected. I had this issue on Windows some time ago long before using Etcher though so this might be related, but not a direct consequence

FetzenRndy, 2019-01-21:
Let me know if you need any informations :)